prusa3d/PrusaSlicer #13449: Choose default action when sending a print to Prusa Connect

ChrisHeerschap, 2024-09-30:
**Is your feature request related to a problem? Please describe.**
New options for sending to the printer chooses the default that rarely works for me.
Previously sending to PrusaConnect the only option was "Set ready and print now" and there was a request on sleekplan to have options. Now there are four options, but the default of "add to print queue" is rarely what I want.

**Describe the solution you'd like**
Be able to choose the default action that works the best with our workflow.

**Describe how it would work**
Either through preferences or via a button on the submit to printer dialog, allow the user to mark one of the four options as the default.

**Describe alternatives you've considered**
The only alternative is to manually choose the option that I want each time I send a print.

**Additional context**
[image]
Now that we have a choice in what to do upon sending the file to the printer via PC, the default has been chosen for us, and it's not what I want.

kylehakala, 2025-01-03:
I came here just for this feature request—glad to see someone else has been wanting this too! 

It would be nice if it were a default that could be set in the Printer presets so that it could be printer-specific.

ChrisHeerschap, 2025-01-04:
For a while, it was defaulting to Set ready and print now, but a couple weeks ago it reverted back to Add to print queue, so it is once again forcing me to do extra mouse clicks that changing a default would avoid.

Acidham, 2025-03-10:
Same here...glad that I am not alone. PrusaSlicer always adds prints to the queue instead of printing now.

kylehakala, 2026-09-19:
I'll check this out on the next release of v3.0.0 alpha--I've been having difficulties with the alpha on macOS.

Jan-Soustruznik, 2026-10-21:
A similar feature request already exists. This request has been merge to keep topics organized, so please continue the conversation in discussion #16134.

closed 2026-10-21 by Jan-Soustruznik