prusa3d/PrusaSlicer #13450: Crash/hang changing a modifier shapes infill percent to 100%

seanmcne, 2024-09-30:
### Description of the bug

I experienced a hang which hung the ui thread making it impossible to proceed without forcing a crash.  

[image]

### Project file & How to reproduce

The hang happened when changing a component/part of an object to 100% infill which produced an exception saying it couldn't be 100% and the UI immediately hung afterwards (I did capture a memory dump, if you have a secure way to transfer it or if you want me to run some commands to pull out any dump information on my end and share it with you).  

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1 on windows

### Operating system

windows 11 23H2 OS Build 22635.4076

### Printer model

prusa i3 mk3s+ latest firmware

u89djt, 2024-09-30:
They'll definitely want the project file as explained in the bug report guidance text.

seanmcne, 2024-09-30:
That's fair - though I've experienced this with other models. I'd rather not distribute it to the public (at least not right at this point), not a massive deal but I'd prefer to share privately just now.  If I have a few free minutes I'll attempt a repro with another project.

seanmcne, 2024-09-30:
@u89djt After realizing it wouldn't take that long, I just retried and easily reproduced the crash - no project file needed and it's even easier than I thought (I'm able to reproduce it 100% of the time now without any project): 

Repro: 
1. Start new project in 2.8.1
2. Add 3 new shapes one at a time (box, cylinder, sphere) 
3. Merge shapes 2 & 3
4. Right click shape 2 (cylinder), click Add setting, check infill box, press Ok 
5. Change shape 2 to infill modifier (from the menu, then check the box for fill density) 
6. Change infill density to 100% and tab off immediately (don't do anything else) 
7. Now you will get an error message press Ok 
8. Immediately click on the shape window (right side where shapes/parts are listed) 
9. Click around try to move the build plate, nothing works, it's a hang 
10. then a crash occurs 

[image]

u89djt, 2024-09-30:
I think the exception you referred to is the standard error message dialog that pops up when selecting 100% infill for patterns that can't be used at 100% density. You'll see that it changes to "rectilinear" usually. The simplest scenario I can construct to reproduce a crash after the 100% error dialog is
1) right click the plate, Add Shape->Box (or any other)
2) right click the box in the object list and choose Add Settings->Infill, then tick the "Fill Density" checkbox
[image]
[image]
The crash doesn't occur if you right click the object and choose the top level "Infill" menu position, which adds a pattern selection box as well as one for percentage. If you do the next steps using that instead, you'll see the infill pattern is changed to work.
3) select the "infill" line in the object list
4) delete 15% and type 100 then - as you described - do not press enter or interact with any menus, but hit TAB
[image]
The error dialog pops up, so hit OK
[image]
Shortly after this, the app crashes/closes
Here's the project file in which you need to type 100 and hit TAB
[infill100tab.zip](https://github.com/user-attachments/files/17196184/infill100tab.zip)
Windows 11, PS 2.8.1

u89djt, 2024-10-01:
...and if you set the infill pattern in the settings tab before you do that, it doesn't crash.
[image]

seanmcne, 2024-10-01:
@u89djt Hopefully, if you follow the repro steps, you should be able to repro.  I know if I jump to step 6 or take another path it doesn't repro, the first steps seem to be important.  If I do not follow the steps exactly like they are below, I cannot reproduce the issue.  I can't speak to other steps that result in a crash (at least not in this bug report).  What I do know is: I can reproduce the crash if I follow all the repro steps, there are probably many other steps which result in **no repro** and maybe more ways to repro but this is all I've got right now. 

I am aware of several workarounds that can be employed (including using 99% or setting the modifier to rectilinear, etc) some of these help in the scenario I was in and others do not apply but I did find a workaround. I appreciate your helping out with offering additional workarounds and context.  Ultimately, I am reporting this as a bug and a bug that caused me to lose quite a lot of work in the slicer. It's also a bug that's reminded me I need to press Ctrl+S after every change I make.  The ultimately goal of this bug report is to hopefully have it fixed, in the near future so others don't experience the crash like I did. Personally, I think all crashes are bad and should be fixed and I'm hopeful these repro steps will help others with a debug environment figure out why there's a crash. 

Thanks!

Repro Steps:
1. Start new project in 2.8.1
2. Add 3 new shapes one at a time (box, cylinder, sphere)
3. Merge shapes 2 & 3
4. Right click shape 2 (cylinder), click Add setting, check infill box, press Ok
5. Change shape 2 to infill modifier (from the menu, then check the box for fill density)
6. Change infill density to 100% and tab off immediately (don't do anything else)
7. Now you will get an error message press Ok
8. Immediately click on the shape window (right side where shapes/parts are listed)
9. Click around try to move the build plate, nothing works, it's a hang
10. then a crash occurs

u89djt, 2024-10-01:
I found a simpler setup that causes the crash - see above. It seems the problem is caused by the need to switch to rectilinear infill at 100% and somehow that isn't done when the infill setting is used on its own as a modifier without the infill pattern setting. Your sequence gets to that place eventually, but the extra steps aren't necessary.

seanmcne, 2024-10-01:
> I found a simpler setup that causes the crash - see above. It seems the problem is caused by the need to switch to rectilinear infill at 100% and somehow that isn't done when the infill setting is used on its own as a modifier without the infill pattern setting. Your sequence gets to that place, but the extra steps aren't necessary. Perhaps interesting: hitting the screenshot key combo immediately after having typed 100 also causes the crash after the error message.

Oh, wow, that's quite strange.

Jan-Soustruznik, 2024-10-04:
Hello, @seanmcne @u89djt , thank you for your report and investigation. I can confirm that we can reproduce crash also on our side. I created internal ticket SPE-2508. We will try to fix in some of new releases of PS. Thank you, have a nice day.

lukasmatena, 2024-11-21:
Fixed in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

closed 2024-11-21 by lukasmatena