prusa3d/PrusaSlicer #13451: The tool for blocking supports doesn't work

Apr3x, 2024-10-01:
### Description of the bug

I want to print a pumpkin with eyes and mouth including teeth. I need supports only in the eyes and mouth. First there is no option to do the support only in the object. Using the tool enforce supports creates support also on the bottom. Using the combination of creating support and blocking support between bottom and mouth has no effect: The support still appears between bottom and mouth and eyes. 

### Project file & How to reproduce

[pumpkin.3mf.zip](https://github.com/user-attachments/files/17199798/pumpkin.3mf.zip)
[Archiv.zip](https://github.com/user-attachments/files/17199811/Archiv.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.1+MacOS-arm64

### Operating system

MacSonoma 14.6.1

### Printer model

Prusa MK4 InputShaper

neophyl, 2024-10-01:
When you define areas to support or not support you are defining areas On The Model geometry that you do and do not want to be supported.  That's it.  A blocker is not and never has been a 'keep out' area.  Once areas are defined then the normal support algorithm takes over and builds the supports based on your other settings.  
You cant just place a blocker modifier below an area that you have defined above as needing support.  It will always build 'downwards' to whatever is below as it needs to.

This is why painting areas for support and blocking is much more intuitive of the process. 

Might I suggest something like in the attached project.  There's a couple of options included but you might also want to play around with the orientation.  
(Note - Also I had to modify the mk4 printer profile to remove a variable before it would let me slice your project so you will need to reselect your normal printer after loading the 3mf.)
[pumpkin_neo.zip](https://github.com/user-attachments/files/17200773/pumpkin_neo.zip)

Apr3x, 2024-10-01:
Thank you very much for your help. The term ‘blocker’ naturally suggests that something is actually being blocked. And if I remember correctly, this is the case with Cura or there is a setting to only create support structures within the object. I was therefore looking for a solution for my MK4, which is not supported by Cura.

I will test the idea of painting on. But even better, of course, is your idea of turning the object so that it can be printed better. Many thanks also for the project file! I wish you all the best and have a great time!

closed 2024-10-01 by Apr3x