prusa3d/PrusaSlicer #13452: Exporting file from Fusion 360 to PrusaSlicer don't work anymore if PS is already open ?

grossmannjean, 2024-10-01:
### Description of the bug

Exporting file from Fusion 360 to PrusaSlicer don't work anymore if PS is already open
I have to close PS, then export from F360 : then PS opens with the object to print. With this new behavior, it seem not possible to add a body on the PS plate.
Is it a bug, or a preference ?
Thanks,
::jean

### Project file & How to reproduce

1. Open PS
2. In F360 utilities, select a body and send it to PS in the interface
3. PS appears with an empty plate !
4. If PS was not previously opened, it appears with the body !

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.1+MacOS-arm64

### Operating system

Mac OS Sonoma 14.6.1

### Printer model

Prusa Mini+

u89djt, 2024-10-01:
Data point from a Windows 11 user: I checked just now and Fusion exports OK the way it always has done for me when there's already an instance of prusaslicer open: it opens a fresh instance of the slicer with the exported body/bodies, leaving the previously opened instance empty. That's via the utilities view, or right click and save as mesh with the print utility drop down menu option selected. That's the way it's always worked for me under Windows, I think.

aeneasw, 2024-10-02:
I have the same issue with my Mac and can confirm that!

macchinette, 2024-10-02:
Hi, in the Autodesk news (link below) it says it's a bug that they fixed.
We fixed an issue where Fusion was not sending parts to slicer via print utility if the slicer was already open in Mac OS.
https://www.autodesk.com/products/fusion-360/blog/september-2024-product-update-whats-new/

grossmannjean, 2024-10-04:
Great, thanks for the information !::jean

https://www.autodesk.com/products/fusion-360/blog/september-2024-product-update-whats-new/

grossmannjean, 2024-10-04:
It came from Fusion 360.
Solved in version (v.2.0.20460) – October 1, 2024

closed 2024-10-04 by grossmannjean