prusa3d/PrusaSlicer #13458: Crash using cut tool

RussellC3D, 2024-10-03:
### Description of the bug

Prusaslicer will crash to desktop when pressing the escape key after editing height value. Does not occur on older versions (2.7.1 | 2.7.4).

Seems to add a question mark and some other numbers to the field right before it crashes. Crash happens on a clean install and all models I've tried.

### Project file & How to reproduce

Load a model >cut tool > click on cut position box > type a value > press escape key > crash

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0 / 2.8.1

### Operating system

Win 11 23H2

### Printer model

Mk3 Mk4 XL

RussellC3D, 2024-10-03:
In earlier versions of the program it would just exit the box and revert value as expected.

u89djt, 2024-10-03:
Confirming reproduced by another win11 user.

TerPuc, 2024-10-03:
Dear @RussellC3D, thank you for your report. The crash was reproduced. 

Issue created: SPE 2507

lukasmatena, 2024-11-21:
Fixed in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

closed 2024-11-21 by lukasmatena