prusa3d/PrusaSlicer #13459: Logging into Prusa Connect through slicer

mskrohn, 2024-10-03:
How do you change what we browser opens to log into Prusa Connect through slicer?  The default generic window it opens doesn't offer stored passwords, so I have to enter it every time I open the program.  It is quite annoying that it either doesn't remember me or allow google password manager.

Please fix!

TerPuc, 2024-10-03:
Dear @mskrohn, for the protection of customer privacy we do not store your passwords. Therefore in case you would like to avoid inserting your password everytime you need to log in, you can link your accounct for example with google as you mentioned. Thank you for your understanding!

m10d, 2024-10-24:
> Dear @mskrohn, for the protection of customer privacy we do not store your passwords. Therefore in case you would like to avoid inserting your password everytime you need to log in, you can link your accounct for example with google as you mentioned. Thank you for your understanding!

@TerPuc - respectfully, this is an unacceptable stance inconsistent with Prusa's promise.  I willingly pay 2x (or more) for Prusa Hardware (presently mk4, and XL5t) because Prusa _promises_ to (and generally seems to) respect my privacy.  I will quote [Prusa's Privacy Policy](https://web.archive.org/web/20241008171652/https://www.prusa3d.com/page/privacy-policy_231258/):
[image]

Requiring me to submit to Google's gathering of my private information & behavior by using their SSO system is in NO way _"only to the extent necessary"_, for a core feature like staying synced in Prusa's App with Prusa's 1st-party cloud.

kocikdav, 2024-11-04:
Dear @mskrohn. Thank you for your comment. 
There are several possibilities for login. Login button will trigger a dialog with an internal browser.
1. You can fill your credentials and login. After you logout and try logging in again, you need to fill your passwords again. There is no remembering of the password and cookies are cleared after login. As you correctly pointed out, we do not process personal data above necessary extent.
2.  Since some users rather uses Google or other accounts, you can choose it instead of filling your credentials. In such case, your default external browser should open. There you login to the 3rd party account and a secure key is shared to the PrusaSlicer app to provide log in.
3. Unfortunately passing such key is rather complicated operation which may fail unexpectedly (f.e. usage of Linux desktop files was a thorn for our side lately.) So as a last resort option, we do open login page for 3rd parties in the login dialog as well. I hope it is clear we do not encourage users to use our browser to log into 3rd party accounts but we do include such possibility as a last resort option. We do clear cookies of our internal browser after each login and do not store passwords anywhere.

I hope this information is helpful to you. If any of the described above seems to differ from your experience, please let us know with some more information (f.e. your OS) so we can start debugging process.

closed 2026-09-29 by prusaslicer-bot[bot]