prusa3d/PrusaSlicer #13460: When changing filament or pausing, push bed to FRONT, not back.

BitSmith99, 2024-10-04:
### Description of the bug

When pausing the print, the bed is moved to the back of printer.  If we want to adjust something, add a magnet or otherwise work with the print, it must be accessible.  Pushing the bed back makes it difficult to reach parts.  
PUSH THE BED FORWARD so we can work with the print.

### Project file & How to reproduce

See Description.
[Wan Scraper Cover.zip](https://github.com/user-attachments/files/17251575/Wan.Scraper.Cover.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.1+win64

### Operating system

WinDoze 10 Home

### Printer model

MK4: 6.1.3+7898

neophyl, 2024-10-04:
This is a function of the printer firmware and not the slicer.  The slicer just commands the printer to pause.

This request needs to go in the repo for your printer firmware, in your case as its a Mk4 that would be here https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/

However this existing one requesting the position be configurable would already seem to cover it. 
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3317

TerPuc, 2024-10-04:
Dear @BitSmith99, thank you for your request. The reply from @neophyl is correct. You can contact our colleagues from firmware with this request. I am closing the issue here as it is not request for slicer. Thank you for your understanding.

closed 2024-10-04 by TerPuc

BitSmith99, 2024-10-04:
I'm not sure if I posted or not, I'm new to GitHub.  The post says "closed"
but seems to have taken my comment.
Please advise?
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3317

Thanks for your help!