prusa3d/PrusaSlicer #13461: Add Settings -> Skirt and brim, is missing skirt.

jmckeown2, 2024-10-04:
### Description of the bug

When adding settings to an object, the "Skirt and brim" options do not include options for skirts. It would be good to be able to set a draft shield on individual warp-prone parts; potentially even different heights. But if it is not intended to allow skirts on individual parts then the option "Skirt and brim" is misleading and should be just "Brim"

### Project file & How to reproduce

1. Add object(s) to project. 
2. Select object, Right click 
3. Add Settings -> Skirt and brim. 

There are no skirt settings. 

[image]
[Skirt_and_brim.3mf.tgz](https://github.com/user-attachments/files/17260183/Skirt_and_brim.3mf.tgz)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1 macOS

### Operating system

MacOS 14.6.1

### Printer model

MK4

neophyl, 2024-10-04:
I'm pretty sure it takes its name from the main section of the settings.  As that sections title is Skirt and Brim and does contain settings for both then it is not practical to change it without a lot of refactoring.

jmckeown2, 2024-10-04:
> I'm pretty sure it takes its name from the main section of the settings. As that sections title is Skirt and Brim and does contain settings for both then it is not practical to change it without a lot of refactoring.

I was presuming it would just be pointing to a different I18N string depending on which UI was getting built.

closed 2026-09-29 by prusaslicer-bot[bot]