prusa3d/PrusaSlicer #13463: additional layer on sliced model, not present in the input STL

el-bart, 2024-10-04:
### Description of the bug

slicing a model with flat top surface:
[image]

results in gcode file with extra layers on sides:
[image]

### Project file & How to reproduce

extra the archive and start `./run`, to get the output model.
[extra_layer.zip](https://github.com/user-attachments/files/17262737/extra_layer.zip)

compare the top most layer in sliced output vs. the input STL

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929 

### Operating system

Debian/12

### Printer model

Voron 2.4 with stealthburner

u89djt, 2024-10-04:
(from a fellow user recognizing an old problem and cheking your model) There's are two corresponding steps in the top surface of your stl, and these take the levels either side just higher and just lower than half way up the layer height. They're only 0.00011027 mm (copied and pasted from Fusion :) ), so not surprising that was missed.
Neither of the upper surface heights in your screenshot are where you want the top surface. The middle section is half a layer too low, and the outer regions are half a layer too high. You can change layer heights strategically to make it top out at the height you want. Happy to talk through.
You might be thinking something like "but why doesn't the slicer stop at whatever height the model specifies." I'm not involved, so I don't have a definitive answer, but I imagine it has to do with the other millions of models the slicer has to cope with. You can choose settings to have it do what you need.

el-bart, 2024-10-05:
thx @u89djt for spotting - u'r right. after adding object intersection to cap-off at a given height (i.e. lowering whole model by a fraction of a mm), the problem went away.

judging by the outcome, it looks like the problem is in algo used to determine "shall i add 1 more layer?". it's apparently only monitoring threshold ("is there any part of the model above current height?") - if so, new layer is added. it's simple, but prone to such edge cases.

the way to solve it would be to change the algorithm's approach to instead optimize for minimal error between the model and actual layered (g-code) structure. i.e. "would adding new layer be more quantization error vs. not adding it?". the analysis can be made on per "island" basis. this way an optimal stop condition can be created, regardless of the actual model shape.

does not sound like a high prio, through. it's a rare case, indeed.

u89djt, 2024-10-05:
Your model is 22.9mm high and you've chosen a layer height of 0.2mm, so the top is hovering around the middle of the highest extrusion layer that might be produced. If it's above half way, material is laid down.
If you chop it off, your print will be 22.8mm high.
If you want it to be 22.9mm high, you need to, for example, set the final layer height to 0.1mm, or maybe a couple of layers there or in the middle to 0.15mm so the object ends at a boundary between intended layers.

u89djt, 2025-05-11:
Same issue: that surface is almost exactly half way up an extrusion layer and it wiggles either side slightly. There's no bug here, so the issue can be closed.
You need to select the layer heights such that the surface you want can be created. It's not possible to create a surface at 1.7mm when all the layers are 0.2mm.
[image]

u89djt, 2025-05-11:
It isn't flat. For example here's a point below 1.7:
[image]
And here's a point above 1.7:
[image]
That's why you have an extra layer there.

u89djt, 2025-05-11:
This is also flat:
[image]

u89djt, 2025-05-12:
Right, and the measurement routines try to find regions that look like flat surfaces by marching along the mesh.
The extrusion generation process looks at each horizontal slice because that's where the extrusions go, and it decides whether to extrude or not based on whether the slice is more or less than 50% occupied.
In theory that could change with non-planar slicing, but as you may have noticed, that is not mainstream (yet?) because only the viral special cases work nicely so far. Variable *width* extrusion is normal now, but when you start moving the tool up and down dipping in and out of the plastic stream, that's presumably a different kind of problem.
You can be certain of where the top of an extrusion is. To get your flat surface at 1.7mm, you need to change one or more layer heights at or below that layer. The minor undulations won't change the occupancy decisions if you line the top up with a boundary between layers.

el-bart, 2025-05-12:
i put some though into the problem as well, and i also do not see anything that'd not require non-planar slicing and that'd not have similar edge cases, just for different models.

i'm closing this ticket, as it's not really a bug - it's just a weir edge case of an otherwise reasonable approach.

closed 2025-05-12 by el-bart