prusa3d/PrusaSlicer #13464: overlaping models when using "duplicate" option with support_material_buildplate_only=1

el-bart, 2024-10-04:
### Description of the bug

when enabling `support_material_buildplate_only=1` and adding `--duplicate 2` command line option, resulting objects overlap, causing segv from slicer:
```
$ ./run 
10 => Processing triangulated mesh
20 => Generating perimeters
30 => Preparing infill
45 => Making infill
65 => Searching support spots
69 => Alert if supports needed
70 => Generating support material
Error: Potentially lost branch!, critical: 1
89 => Calculating overhanging perimeters
88 => Generating skirt and brim
[2024-10-04 18:58:14.428396] [0x00007f866e526d40] [error]   gcode path conflicts found between model.stl and model.stl
90 => Exporting G-code to model.gcode
Segmentation fault (core dumped)
```

looks like outside-of-model supports are not taken into account when calculating object size.
[image]

### Project file & How to reproduce

download the model and config:
[colliding_objects.zip](https://github.com/user-attachments/files/17262816/colliding_objects.zip)
call `./run` to build. observe segv. check output gcode.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929

### Operating system

Debian/12

### Printer model

Voron 2.4 with stealthburner

neophyl, 2024-10-04:
No they are not.  This has been reported multiple multiple times and a warning was added to the gui version.  But the user still has to space the objects further apart.  Either that or you merge all the objects into 1 object and the supports will then take that into account.  This is the intended operation at the moment.
I wouldn't expect the command line to be any different.  

For example
https://github.com/prusa3d/PrusaSlicer/issues/10464
https://github.com/prusa3d/PrusaSlicer/issues/10451
https://github.com/prusa3d/PrusaSlicer/issues/10376
https://github.com/prusa3d/PrusaSlicer/issues/10032
https://github.com/prusa3d/PrusaSlicer/issues/9861
https://github.com/prusa3d/PrusaSlicer/issues/9860
https://github.com/prusa3d/PrusaSlicer/issues/9835
https://github.com/prusa3d/PrusaSlicer/issues/9685
https://github.com/prusa3d/PrusaSlicer/issues/9517
https://github.com/prusa3d/PrusaSlicer/issues/9510
https://github.com/prusa3d/PrusaSlicer/issues/8012
(list taken from a previous issue, now closed)

el-bart, 2024-10-05:
ah - indeed. looks like #316 is the 1st time it's mentioned. not sure how i missed that yesterday... :face_with_diagonal_mouth: thx @neophyl!

i'll close this as a duplicate, and just mention the new part: the SEGV issue + wrong auto-positioning when `--duplicate`ing.

closed 2024-10-05 by el-bart