prusa3d/PrusaSlicer #13466: high cpu usage when idle after startup

jiri-jirus, 2024-10-06:
### Description of the bug

prusa-slicer exhibits very high cpu usage after startup, with no model loaded.
[image]

### Project file & How to reproduce

Start Prusa-slicer 2.8.1 and watch CPU usage
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 10

### Printer model

Prusa MK3S+

fe60, 2024-10-07:
Duplicate of #13127

TerPuc, 2024-10-10:
Hello @jiri-jirus, thank you for your report. I will close this issue as duplicate. You can follow the webview issue development here: #13127. Have a nice day!