prusa3d/PrusaSlicer #13468: Allow setting the speed of the FIRST layer in the wipe tower

skoblins, 2024-10-07:
**Problem**
I can set the first layer speed but this doesn't affect the wipe tower. The wipe tower travels in the middle of the print, because of this.

**Solution**
The first layer speed should also affects the wipe tower.
[image]

u89djt, 2024-10-07:
(fellow user following the hub) first layer speed does set the speed of the tower's print. The very first squirt to clear the nozzle proceeds at full speed - because it has to to perform its job, I assume? But the box is printed according to the first layer speed setting.
If we look at the data you have providede - that single image - you obviously have a problem with warping. If you can follow the submission instructions - include all the information requested and perform the due dilligence - someone might take a look. There are threads in the forums about dealing with warping.

skoblins, 2024-10-08:
> (fellow user following the hub) first layer speed does set the speed of
> the tower's print.
No, it doesn't. The wipe tower first layer speed is completely out of
control.

> The very first squirt to clear the nozzle proceeds at full speed - because
> it has to to perform its job, I assume? But the box is printed according to
> the first layer speed setting.
No it's not, it's printed much faster, and I don't refer to puring, I refer
only to the first layer,  which is one of the most important things during
the print - no matter if it's the wipe tower or not. When this  is done to
fast, the print is likely to fail at later stages. And I don't complain
about the speed of the remaining layers, they are fine.

> If we look at the data you have providede - that single image - you
> obviously have a problem with warping.
Yes, but I deal with warping ideally in the MAIN object (same print). *The
main object has no problems, because the machine does what I tell it to do*.
I have lowered the first layer speed - to be more precise: the
acceleration, and it sticks very well to the surface.
The wipe tower however, is out of control, and that's why I created this
ticket.

> If you can follow the submission instructions - include all the
> information requested and perform the due dilligence - someone might take a
> look.
The test is very simple. Use cheap PLA, lower the acceleration / speed of
the first layer by e.g. a factor of 4, and just watch what the printer is
doing with the first layer of the wipe tower.

> There are threads in the forums about dealing with warping.
Yes, thanks, I deal with problems well as long as the program listens to
what I tell. However, In this case it is trying to be smarter than me,
otherwise I would not bother with creating tickets.

u89djt, 2024-10-08:
So to be clear, where you said speed in your report, you didn't mean speed, you meant acceleration? And by inference, each reference to fast is to high acceleration.
I see that the first layer speed of the wipe tower is affected by the first layer speed setting.
Here are two boxes at the default speeds:
[normal.zip](https://github.com/user-attachments/files/17292264/normal.zip)
[image]
Here are those boxes with a much lower first layer speed:
[2mms.zip](https://github.com/user-attachments/files/17292273/2mms.zip)
[image]
So the first layer speed is used. It's not completely out of control.
Now we know you mean acceleration, we can test that and see that the first layer acceleration setting is not used in the first layer of the wipe tower:
[125mms2.zip](https://github.com/user-attachments/files/17292288/125mms2.zip)
[image]
The speed reached in the main run of each extrusion is the same, but the acceleration and deceleration are lower.
The indication of the necessity of including a project file and screenshots of circumstances is there to bypass the need for investigation in the space of deficits in a user's knowledge or articulation. These accrue and develop over time. For example, now you know that you can perform some investigation of acceleration without needing to perform a print, and that acceleration and speed are distinct quantities treated differently by the slicer.
Acceleration is indeed not accessible. Your report did not indicate explicitly, or implicitly and accessibly in firmly requested data, that this was the problem you were facing.
I'll take this opportunity to re-emphasize that I am just a fellow user, not affiliated with Prusa. I get to be blunt.
At this point, if we're lucky, someone at Prusa might step in and explain *why* acceleration isn't accessible. My amateur guess is that acceleration doesn't make much difference to warping in the first layer because the squish at the reliable hot surface of the print bed dominates the flow behaviours, and the wipe tower is just using up time. But I don't know, and may be comically wrong.
I would enjoy hearing about it from them :)

skoblins, 2024-10-08:
You are right. Let's then use the term: ACTUAL speed.
I found lowering the acceleration a very efficient way of improving stickiness. It slows down at the details, and speeds up at more straight curves, so it's often and ideal tradeoff. In the attached screen you can compare my actual speeds of the 1st layer in my project. The max actual speeds seem to reach the target value (40mm/s). The accel is set to 30mm/s2.
[image]

u89djt, 2024-10-08:
While you're editing your original post, add the project file(s). You get to act as a good example to folk looking around at issues on the hub.

u89djt, 2024-10-08:
OK, continuing due diligence efforts this side of the fence: there's something related here: https://github.com/prusa3d/PrusaSlicer/commit/8e9251f2f0044bdbe6e2985cbc6b30b0384410c7
Coming from https://github.com/prusa3d/PrusaSlicer/issues/10854
So my value probably runs out here, and @lukasmatena might call it?

seancmonahan, 2024-10-16:
I use a post-processing script I published as a gist that clamps all travel move speeds and all accelerations on the first layer. I have not tested it yet to see its effect on wipe tower printing moves, but it should at least clamp the first layer acceleration for it. Please let me know if it works for you.

https://gist.github.com/seancmonahan/05f92505162a7081b1539686ad33f535

Usage is to put in Post-processing scripts:
`{full-path-to-this-file} {your-desired-first-layer-travel-speed-in-mm-per-second}`

It's hacky as heck. You pass it your desired limited first layer travel speed. It reads the first layer speed, the first layer acceleration , and the travel speed from environment variables that PrusaSlicer exports to it (e.g. `SLIC3R_FIRST_LAYER_ACCELERATION`).

(On Windows, I think it would be
`{path-to-your-python-interpreter} {full-path-to-this-file} {your-desired-first-layer-travel-speed-in-mm-per-second}`
)

(On Linux (and possibly macOS), make sure to mark the file executable. `chmod +x limit_first_layer_travel_speed.py`)

closed 2026-09-29 by prusaslicer-bot[bot]