prusa3d/PrusaSlicer #13472: Printer profile request

Parogers90, 2024-10-08:
Good Afternoon,

I would like to request that you add a profile into prusaslicer for the following printer:

Creality ender 3 V3 SE

I would like to use your slicer software and have heard nothing but positive user experiences from friends, however I cannot get in on this because there is currently no profile for my printer.

Please take this into consideration.

Thankyou.

u89djt, 2024-10-08:
I've only got prusa printers, but I googled and found https://www.makersmuse.com/ender-3-v3-se-prusaslicer-2-7-0-profile
Look here, for example, for a clarification of responsibilities for third party profiles: https://github.com/prusa3d/PrusaSlicer/issues/12550

Parogers90, 2024-10-25:
> Hi @Parogers90 <https://github.com/Parogers90> , to make the profile a
> permanent resident of your install of the slicer by, you can:
> Open the example 3mf file you downloaded - that temporarily loads the
> profile
> Go to the printer tab and hit the save icon
> You probably saw that If you try to save, it will tell you that you can't
> overwrite an external profile, so
> Rename it (I deleted everything except what's in the brackets) and hit save
> And it's now installed in your slicer.

Evening,

So far so good. Printing a calibration cube to test it out. Thanks for your
help.

u89djt, 2024-10-25:
Great! Thanks for checking in.

Jan-Soustruznik, 2025-08-29:
closed as duplicate of  #14797