prusa3d/PrusaSlicer #13474: In vase mode, different(color of) speed is displaying all over the place

5jvm0u4, 2024-10-09:
### Description of the bug

As mensioned, it seems to be just a visual bug, no actual speed changes occored in the printing process.

### Project file & How to reproduce

Load the provided .3mf(zip) file as a project, it's reproducible, at least in 2.8.1
[bug.zip](https://github.com/user-attachments/files/17304363/bug.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1+win64

### Operating system

windows 11 23H2

### Printer model

Voron 2.4R2 with self-designed toolhead

u89djt, 2024-10-09:
(Intrigued fellow user going for it)
So we have implied speed variations smaller than the precision of the labels in the speed display, which is showing identical labels on each line. Most of the speeds are in the middle of the range with brief flicks to one or the other extreme. I'll start with a minimum-inputs-from-boot route to the bug, then use the original example to maybe get an idea of the size of numerical glitches.
The shortest path I've found so far from an empty slicer to seeing red and blue flecks in the speed display is as follows:
Slicer opens to MK3S 0.2mm speed profile
Add cylinder to bed
Make the following settings to instruct the slicer to output a single wall with no top, bottom, infill, brim or skirt, not in vase mode:
infill=0%
perimeters=1
bottom layers=0
top layers=0
skirt loops=0
Delete the purge line by removing its extrusion commands from the custom gcode. I did this one in the MK3S profile, so I deleted lines
{if filament_settings_id[initial_tool]=~/.*Prusament PA11.*/}
to
{endif}
Now slice and switch to speed view to get this:
[minimumchangesspeedglitch.zip](https://github.com/user-attachments/files/17310843/minimumchangesspeedglitch.zip)
[image]
And if you need more convincing, switch on fuzzy surface to get this:
[image]
Playing with the settings in OP's project shows that the numerical glitching is something like 0.000001mm/s either side of the settings value.
[image]
Use OP's project file. You have to set the first layer speed yourself to 99.99999 because it is rounded up to 100 on saving so I can't post it with the value already filled in.

That's 99.99999 with five nines after the decimal point.
Very approximately, the first layer runs at a nominal 99.99999mm/s with jumps to about 99.999995 and 99.999985. The main body runs at 100 with jumps of again approximately 0.000005.
0.000005 in 100 is 5e-8 so wild speculation wonders about something with 24 bits being involved, but that's just a random computer scientist performing very superficial pattern matching :)
Here's the actual speed display from the same slice:
[image]
Over.

u89djt, 2024-10-10:
Neat - looks like something eminently traceable:
[boxneatglitches.zip](https://github.com/user-attachments/files/17330822/boxneatglitches.zip)
[image]

u89djt, 2024-10-10:
@5jvm0u4 I dig that you found this. If this isn't effecting/reflecting disturbances in the actual gcode output, and I can't find any evidence that it is, since no-one will ever see this effect in a practical print job, I predict a "closed as not planned" in 3... 2... 1...

u89djt, 2024-10-10:
If I vary the acceleration cap, the glitches move around, so true speed values are making their way into display decisions/values they shouldn't. It seems like a bad idea to leave that kind of crossover lying around in code.
[smearedglitches.zip](https://github.com/user-attachments/files/17334252/smearedglitches.zip)
[image]

RomanLesovoi, 2024-11-07:
is this problem?
Gcode contains frequent speed changes in vase mode. at version 2.7.2 gcode clean and speed don't change.
At print model all round surfaces look broken in different places like from many lines, not a perfect circle
[image]

> G1 X170.54 Y66.795 Z98.017 E1536.66138
> G1 Z98.019 X170.44 Y66.145 E1536.68817
> G1 F8382.146
> G1 X170.369 Y65.491 Z98.021 E1536.71497
> G1 F8396.848
> G1 X170.337 Y65.014 Z98.023 E1536.7345
> G1 F8369.938
> G1 X170.311 Y64.176 Z98.026 E1536.7688
> G1 Z98.028 X170.326 Y63.519 E1536.79557
> G1 F8379.658
> G1 X170.369 Y62.863 Z98.03 E1536.82239
> G1 Z98.032 X170.44 Y62.209 E1536.84925
> G1 F8397.958

u89djt, 2024-11-07:
That looks like a different phenomenon. Go ahead and post that project file. You need to zip it to post it.

RomanLesovoi, 2024-11-07:
> That looks like a different phenomenon. Go ahead and post that project file. You need to zip it to post it.
any project, just make cylinder in vase mode for example. if i make gcode in 2.7.2 print model look nice, new version not and in ActualSpeed i see defferent speed

u89djt, 2024-11-07:
Post your project, because there are many things that can differ between setups. You may have things going on in your defaults that make it behave differently to mine. When you're examining an issue, you need to be certain you're looking at precisely the same thing. The way we achieve that here is by sharing project files. Save the 3mf, zip it, and post it.

neophyl, 2024-11-08:
I asked for the project on their forum post too. https://forum.prusa3d.com/forum/prusaslicer/is-this-problem-or-bug-or-my-broken-hands/#post-726509

closed 2026-09-29 by prusaslicer-bot[bot]