prusa3d/PrusaSlicer #13475: Prusa Slicer 2.8.1 crashing on startup under Windows 10

Herrcookiekiller, 2024-10-09:
### Description of the bug

When starting Prusa Slicer 2.8.1 it shows the startup splash image, but then goes unresponsive and crashes. Prusa Slicer 2.8.0 worked fine for months now, but i decided to update and cant get it to work now.

[image]

I already did a complete reinstall.
I already tried starting as admin.
I already restarted my computer.

Here is the Windows Event Report
```
- System 

  - Provider 

   [ Name]  Application Error 
 
  - EventID 1000 

   [ Qualifiers]  0 
 
   Version 0 
 
   Level 2 
 
   Task 100 
 
   Opcode 0 
 
   Keywords 0x80000000000000 
 
  - TimeCreated 

   [ SystemTime]  2024-10-09T12:17:49.1319570Z 
 
   EventRecordID 16469 
 
   Correlation 
 
  - Execution 

   [ ProcessID]  0 
   [ ThreadID]  0 
 
   Channel Application 
 
   Computer DESKTOP-409IOVA 
 
   Security 
 

- EventData 

   prusa-slicer.exe 
   2.8.1.0 
   66eaddda 
   EmbeddedBrowserWebView.dll 
   121.0.2277.83 
   65b1bfd9 
   80000003 
   00000000003450d0 
   25a4 
   01db1a4540633993 
   C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe 
   C:\Program Files (x86)\Microsoft\EdgeWebView\Application\121.0.2277.83\EBWebView\x64\EmbeddedBrowserWebView.dll 
   28340ac2-f4e7-4b2a-9fd7-4c6fab5b8c0b 
```

### Project file & How to reproduce

Open Prusa Slicer 2.8.1 under Windows 10 Pro 22H2

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1.0

### Operating system

Windows 10 Pro 22H2

### Printer model

MK4S

derelictjet, 2024-10-16:
I have this same issue since about a week ago (I think there may have been a windows update?). It's not just 2.8.1 either. I tried downloading a copy of 2.8.0 and it crashes on startup too.

Herrcookiekiller, 2024-10-17:
> I have this same issue since about a week ago (I think there may have been a windows update?). It's not just 2.8.1 either. I tried downloading a copy of 2.8.0 and it crashes on startup too.

I uninstalled 2.8.1, then reinstalled the latest 2.8.0 and the 2.8.0 version didnt crash for me. 
I also havent done any windows updates since i installed 2.8.0.

It honestly feels like the devs dont care, from what i can tell it appears to be an Edge problem, since i have completely uninstalled and removed every last bit of Microsoft's Edge and the Windows Event Error says something about "EmbeddedBrowserWebView.dll "

closed 2026-09-29 by prusaslicer-bot[bot]