prusa3d/PrusaSlicer #13481: Warning message on launch

epetit91, 2024-10-13:
### Description of the bug

With release 2.8, I reported a warning message.
This message has still not been fixed with version 2.8.1. I don't know if other people have encountered this problem with this OS.
Is it planned to look at this point?
[image]

### Project file & How to reproduce

Simply launch Prusaslicer 2.8.1 on an iMac macOS 10.12.6

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1 on iMac

### Operating system

macOS 10.12.6

### Printer model

Prusa MK4S

fe60, 2024-10-13:
Same warning as shown in https://github.com/prusa3d/PrusaSlicer/issues/12975#issuecomment-2198659052

TerPuc, 2024-10-14:
Dear @epetit91, thank you for your report. I close this issue as duplicate and I will keep open your original post from June 28th. Thank you for your understanding.

closed 2024-10-14 by TerPuc