prusa3d/PrusaSlicer #13483: Slicer crashes when generating organic supports on this model

rsteckler, 2024-10-13:
### Description of the bug

Slicing this model with snug supports works.
Slicing other models with organic supports works.
Slicing this model with organic supports gets to about 70% slicing, then PrusaSlicer crashes with no error messages.

Project attached.  Just open it and hit slice and it crashes

### Project file & How to reproduce

[big-pumpkin-candy-bowl.zip](https://github.com/user-attachments/files/17357610/big-pumpkin-candy-bowl.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.1+win64

### Operating system

Windows 11

### Printer model

Prusa XL 5 toolhead

murk-sy, 2024-10-14:
Could not reproduce

[image]

Do you have a 13th/14th gen Intel cpu?
I believe this tool by Intel will let you identify if your processor is affected by the recent issues.
https://www.intel.com/content/www/us/en/download/15951/intel-processor-diagnostic-tool.html

u89djt, 2024-10-14:
No problem for me on Ryzen with 4090, either. What do you mean by crash? Does the application close, or does it just sit there for a long time? It's a big processing job, and you haven't said what processor you're running on or how much RAM you have both for the processor and any GPU. If it can't run in core for you, swapping will make it very, very slow.
What happens for you if you scale the model down by 50%?

rsteckler, 2024-10-14:
Thanks for the pointer.  I have an Intel 14900k (problematic intel version) and had updated the bios in late August to Intel microcode 0x129 and assumed I was good.  There was *another* update in late September for Intel microcode 0x12b.  Just applied that update and the issue appears to be fixed.

closed 2024-10-14 by rsteckler