prusa3d/PrusaSlicer #13486: Allow Multiple 3mf files to be dragged and dropped into prusaslicer at the same time

thatguymendel, 2024-10-14:
I have files saved in 3mf that have seam painting, support painting and per object settings, but when you try to drag multiple 3mf files into prusaslicer at the same time, only one gets loaded. This also happens when you try to load an stl and a 3mf file at the same time, only the 3mf gets loaded. 

I understand this happens because a 3mf is a project file and you dont want to try to load multiple slicer settings at the same time, but the slicer settings are only loaded when you load the file with "open project". When you drag the file into prusaslicer, only the object is loaded. 

You are able to load multiple 3mf files at the same time with "add" or ctrl+I. I dont see why dragging files into prusaslicer cant behave the same way. Dragging files into prusaslicer doesnt load the slicer settings anyways

u89djt, 2024-10-14:
Minor correction: when you drag a 3mf in, you get a dialog asking how you want it to be imported. 
[image]

thatguymendel, 2024-10-14:
My bad i forgot i selected import 3d model only and checked dont ask again.

u89djt, 2024-10-14:
If you want it back, you might have to set show_drop_project_dialog = 1 in PrusaSlicer.ini
I haven't found a restoration route via menus in the slicer yet.

u89djt, 2024-10-15:
If you type "dialog" in the search box, you get the option to switch it back on again.

closed 2024-11-20 by thatguymendel