prusa3d/PrusaSlicer #13487: Sequential printing tall objects

luiselse, 2024-10-15:
**Is your feature request related to a problem? Please describe.**
I often print thin tall objects, I would like to print several tall objects in a sequential printing.

**Describe the solution you'd like**
I would like either an additional Y-axis clearance option (instead of radios) that clears an entire Y-axis band instead of just a bubble, or use the existing radios to define a Y-band and allow tall objects as long as they are in different Y-axis bands, enforcing clearance for the bubble in the same Y on small objects, and the entire Y-band for tall objects.

**Describe how it would work**
When placing two tall objects (taller than the clearance height), instead of blocking placement, the slicer would allow them if they are not in the same Y-band. This would enable printing 2/3 tall objects sequentially without collision.

**Describe alternatives you've considered**
Currently, I just change the extruder clearance height to a big number, effectively disabling the check, I then position the objects so they're not in the same Y-band. However, this approach is prone to human error, and I cannot monitor the printer for 8h, so I would prefer to have the slicer validate that objects will not collide with the printer arm.

**Additional context**
I use the MK4, but this issue is common across bed-slinger and XY printers (not the same for "stranger" printers like Delta).

yoyo42, 2024-11-10:
I like the idea of this, especially if the auto-arrange can be tweaked to use it as well. Even without that a manual check would be very useful.

rmpel, 2026-09-18:
I think it wise that this behavior is tested in PS3, and if it is solved, we close. If not resolved, we remove the legacy-issue label and keep open.

Agreed?