prusa3d/PrusaSlicer #13488: Large hole in part ignored (auto-fix issue?)

ebeling-rump, 2024-10-15:
### Description of the bug

This stl file contains a large hole:

[image]

However, when importing into PrusaSlicer, this hole is ignored:
[image]

This can also be seen when performing a cut through the part:
[image]

In other slicer, namely Simplify3D and Cura this seems to work.
[image]

[image]

Could this be related to PrusaSlicer auto-fixing and thereby filling the hole?

I've checked the normals of the hole and they are pointing towards the inside as expected.

[image]

### Project file & How to reproduce

[bridging_test.3mf.zip](https://github.com/user-attachments/files/17376134/bridging_test.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1+MacOS-arm64

### Operating system

MacOS 13.2.1 and Windows 11

### Printer model

Prusa XL 5T

ebeling-rump, 2024-10-15:
Hey Dave,

thanks for looking into this!

Here is the original stl, exported into ASCII format:

[bridging_test.stl.zip](https://github.com/user-attachments/files/17380495/bridging_test.stl.zip)

u89djt, 2024-10-15:
At this point, I'm stumped. Will probably look again at some point.
In the meantime, if you can describe your software and processes chain that got you to your 3mf file, someone might recognize it.

u89djt, 2024-10-15:
If you import your stl into meshmixer and then export it immediately, that new stl imports into the slicer and gives an object with the hole. So meshmixer disliked something in the mesh and fixed it, and made prusaslicer happy, too.
[meshmixerrepair.zip](https://github.com/user-attachments/files/17381782/meshmixerrepair.zip)

ebeling-rump, 2024-10-16:
Very interesting! I could reproduce this behaviour with Meshmixer. 
[image]

When exporting the meshmixer stl in ASCII format and comparing it to the original, it seems like Meshmixer did a few different things:
[image]

1) rounded floating point numbers
2) turned floats into integers wherever possible
3) added indentation
4) changed all normals...

I'm not quite understanding the changed normals. For the first facet all vertices are at the same height z=4. Therefore I'd expect the normal to be pointing upwards, i.e. (0,0,1), but it is pointing into the y-direction with (0,1,0).

u89djt, 2024-10-16:
I guess it ingested the original file into its own binary data structures, and that's it's standard output format? The normals - no ideas yet. At face value (aha) they're nonsensical, but here we are with a working mesh...

closed 2026-09-29 by prusaslicer-bot[bot]