prusa3d/PrusaSlicer #13495: File impossible to slice

renaud-lebrun, 2024-10-16:
### Description of the bug

I have an STL which is not "awfully big" (less than 2 million triangles), however, I can not slice it, PrusaSlicer fails to slice it.

### Project file & How to reproduce

It's more than 25 Mb. However, this 3mf file can be downloaded here : https://entrepot.recherche.data.gouv.fr/dataset.xhtml?persistentId=doi:10.57745/GPWDXP

=> then I exported the plate as an STL (because when opened directly with prusaslicer, the model looked weird and had many "open faces" for some reason) and reimported it in PrusaSlicer (then the STL file had no error), and tried to slice it (without support, 0.20mm Structural, Prusament PLA, Mk4 Input Shaper 0.4 nozzle)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1 +win64

### Operating system

Windows 11

### Printer model

MK4

renaud-lebrun, 2024-10-16:
Many thanks for your answer and your time !

To be honest, I have never seen a 3D model that ugly myself, it's a user at my research lab who asked me to print this. I usually work with heavy (much heavier... sometimes 20 million faces) and ugly broken surfaces extracted from X-ray µCT dataset, and so far PrusaSlicer had never failed to slice any of them. So that is why I was surprised and reported this issue.

Thanks again, I will simplify these models and try to print them that way for my user (he works on wood and would like to create such "wood" models for outreach).

u89djt, 2024-10-20:
Just demonstrating the feasibility of support. The fragments of the sieve plates in the phloem are bad news, but they indicate general location, so they might as well be there.
@neophyl is right - that's not a mesh in the traditional sense :) It's at the very least inside out with triangles that don't match up. I fixed those two factors in meshlab, imported to prusaslicer, simplified to mid, reduced the resolution of the slice and gcode, chopped it down and supported it on its side. Also switched to the classic perimeter generator, because arachne had a number of stray lines in space.
[halffixedsupported211_alisier.zip](https://github.com/user-attachments/files/17450578/halffixedsupported211_alisier.zip)
[image]
Your original 3mf seems to slice successfully in cura and bambu, so if they have profiles for your printer, you're in luck.

closed 2026-09-29 by prusaslicer-bot[bot]