prusa3d/PrusaSlicer #13498: Bridge ancoring issue again

Ma-xik, 2024-10-16:
### Description of the bug

non-ancored bridges 
view from below:
[image]
view from top:
[image]

### Project file & How to reproduce

just slice and see bridges inside holes
[toe1.zip](https://github.com/user-attachments/files/17398007/toe1.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

 2.8.1+win64

### Operating system

win 10

### Printer model

neptune 3 pro + klipper

Ma-xik, 2024-10-16:
i researched this issue better. this problem appear in case when thin part displaced against layers f.e. coordinates of layer 0.4-0.6 and coordinates of bridge should be 0.45-0.65
[image]

dmeehl, 2026-02-19:
@Ma-xik,  I believe the same issue was [fixed in 2.7.3-alpha1](https://github.com/prusa3d/PrusaSlicer/issues/11150#issuecomment-2012546910) but has returned. I [posted a very simple object](https://github.com/prusa3d/PrusaSlicer/issues/11150#issuecomment-3927676000) on that ticket which displays the problem. The problem doesn't exist in 2.7.4 but does in 2.9.4. I'm not sure when it came back, I run linux and was unable to get 2.8.x running for now. 

I know it's been a while since your last comment, but if you still have the model, you could see if you can reproduce in 2.7.4.

Tupson444, 2026-09-19:
close only if resolved