prusa3d/PrusaSlicer #13502: negative volumes not reflected in G-code thumbnails

coderjo, 2024-10-17:
### Description of the bug

Negative volumes added to a part are not rendered properly in G-code thumbnails.

G-Code as exported, in PS' G-code Viewer
[image]

thumbnail as shown in OctoPrint
[image]

### Project file & How to reproduce

I made a 44mm diameter by 2mm cylinder and added a 42mm diameter negative volume cylinder in the middle of it. I then sliced it and sent it to Octoprint. The thumbnail shows a 44mm by 2mm disc, rather than a ring.

[round_hygrometer_spacer_2mm.zip](https://github.com/user-attachments/files/17407465/round_hygrometer_spacer_2mm.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.1+win64

### Operating system

Windows 10

### Printer model

Ender 3 v2 with SKR Mini E3 v3.3; OctoPrint 1.10.2 with Slicer Thumbnails 1.0.7

neophyl, 2024-10-17:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/9046

coderjo, 2024-10-17:
Oops. I didn't think to search for "Modifiers".

closed 2024-10-17 by coderjo