prusa3d/PrusaSlicer #13503: There needs to be an easy way to select visually objects to cancel during the print

jackTfrost, 2024-10-19:
**Is your feature request related to a problem? Please describe.**
I am printing on the XL a full print plate of relatively small objects.  A few of them have not stuck to the bed and I would like to cancel them but there is no easy visual way to do this. The file size is 90mb so I had to load it directly from the USB drive.

**Describe the solution you'd like**
There should be an easy way in PrusaSlicer when viewing the current print to simply select the object and cancel that one.

**Describe how it would work**
I don't know how to describe it any easier. Prusa Connect should allow the manipulation and control of all prints that are pringing and allow the visual selection and cancelation of specific objects.

**Describe alternatives you've considered**
There are not really any alternatives here until Prusa releases an App with which to control the printer and manage the print.

murk-sy, 2024-10-22:
Duplicate of:
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3418
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3744

jackTfrost, 2024-10-22:
> Duplicate of:
>
>    - prusa3d/Prusa-Firmware-Buddy#3418
>    - prusa3d/Prusa-Firmware-Buddy#3744

Thank you!

Tyler

closed 2026-09-29 by prusaslicer-bot[bot]