prusa3d/PrusaSlicer #13507: Prusa Connect still logs me out periodically

quintesse, 2024-10-22:
### Description of the bug

Even with the latest version 2.8.1 I get continuous "Missing or invalid security token" errors in the Prusa Connect tab.

This issue is similar but not exactly the same as #13137, I'm not getting blank pages anymore, but just a "working" page that doesn't let me do anything because it's missing the security token.

I think it would be very useful to have a way to force PrusaSlicer to re-authenticate without having to restart (which forces me to save ephemeral changes to a scrap project just so I don't lose those changes when restarting and then having to delete that file. It's a bit of a nuisance)

### Project file & How to reproduce

Can't give an exact foolproof way to reproduce, but I think it's simply letting PrusaSlicer open for a long time.
If "enough" time has passed this problem will occur (and there doesn't seem to be a work-around except restarting PrusaSlicer).

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

1.8.1

### Operating system

Windows 11 Pro

### Printer model

XL 2 head

m10d, 2024-10-24:
While the result is similar (must re-log in), your `missing or invalid token` error does sound different than the "failed to connect" as seen in https://github.com/prusa3d/PrusaSlicer/issues/13137#issuecomment-2282478057 ; no comment on if they share a root cause.    

Can you please clarify a few things from your OP: (also, edit your title so it doesn't look like a dup  of 13137)

- does re-starting the app result in you logged in (eg is this something like an `oauth` refresh access/bearer token issue, not an actual logout)?  Or do you have to fully re-log-in?
- while you're experiencing this error, does anything on the Connect tab refresh (or is it just displayed, but staying in a stale state?  If it's refreshing, that would imply multiple paths to handle the auth refresh (odd), or a security issue).

Note your symptoms sound a lot like a failure on some paths to request a new `access token` with (assuming restarting the app DOES fix the issue, no password required) the `refresh token` the app stores.  access tokens are ~short lived, they're supposed to expire quickly as part of their design.  A new one is obtained _behind the scenes_ by whatever app, using the refresh-token.  At least that's OAUTH2, if you're curious)

quintesse, 2024-10-24:
@m10d it does indeed seem more like a refresh issue, given the fact that it only happens after a certain time has passed and that,  as you surmised, the app reconnects without needing a login after being restarted.

As to the second point, I did not pay 100% attention, will try to reproduce now and take a closer look. But as far as I can recall the page shows stale information. As you can see from the screenshot it actually showed me as still being logged in (both in the app itself as in the embedded Prusa Connect page). I could also change tabs. But many (all?) actions caused an error popup.

I'm somewhat familiar with OAuth2 and the idea of refresh tokens so it could indeed be that the refresh is not being handled correctly. I don't know how short-lived the access tokens are. If they are in the minutes then it would mean that the refresh normally works but sometimes fails (and then doesn't retry but enters in a failed state that it can't get out of) or the tokens are somewhat longer-lived, let's say an hour, which could possibly mean that the refresh doesn't work at all (at least in my case).

quintesse, 2024-10-24:
I was able to reproduce it by switching back to any other tab and leaving it open on the desktop. When I left it on the Prusa Connect tab it didn't seem to happen. But perhaps minimizing would have worked as well as @michael-gillett found out.
I can still move to different tabs and some will show _some_ information but most will just pop up a bunch of extra errors.
I once tried to make a change (set a hotend temperature) and after a couple of minutes the tab went completely blank with just a single "Connection Failed. Something went wrong" message in the center. Logging out and in again make things work, so does restarting.
(NB I really think we need a way to force a refresh, it's a web page after all :-) )

bryn51, 2024-10-27:
I also observe this issue occurring. The Prusa Connect login button at top right of Prusa Slicer screen shows user is logged in, but when Prusa Connect button is clicked, the details are inaccurate.  When "Set Ready" is clicked on a printer, nothing happens, or sometimes it says "Missing or invalid security token".  What ought to happen is that Prusa Slicer keeps the login status "alive" ongoing, and when "Prusa Connect" button is clicked, if necessary the prusa connection login status is refreshed and the contents of the screen brought up to date immediately.

a1athrop, 2024-10-30:
I am having this same issue.

justinpierson, 2024-11-02:
Same issue here, if I leave PrusaSlicer open for a while and then try to access the printers files I get the error "Invalid Security Token" I can still send prints to it, I just cannot do anything with the printer.

MichaelEhrlichman, 2024-11-05:
Same issue here.

Jan-Soustruznik, 2024-11-08:
It seams similar to https://github.com/prusa3d/PrusaSlicer/issues/13439

quintesse, 2024-11-08:
> It seams similar to #13439

Actually that one seems more similar to a bug that was supposedly fixed in 1.8.1, namely #13137

feesta, 2024-11-08:
I have also been having this problem on 1.8.1 running on MacOS. I tried running from the command line so logs show up. This error popped up after about 37 minutes of PrusaSlicer sitting in the background which corresponds to seeing 'Missing or invalid security token' notifications in the app:

> [2024-11-08 10:45:52.576715] [0x00000001e9038f40] [error]   Failed communication with Prusa Account: 0x60001db75c48 action failed (400): {
>   "message": "key parameter is invalid"
> }

itshanney, 2024-11-15:
+1 to experiencing this issue. It definitely happens after PrusaSlicer has been sitting idle for a while (4+ hours) and you come back to try and do something or check on a print. There is some timeout that it is hitting and then throwing the missing token error.

guiweber, 2024-11-18:
Same issue here with V2.8.1 on CachyOS.

AndreasBackx, 2024-12-01:
I seem to face this issue every time on Linux. Both with the AppImage and the Fedora RPM.

When opening Prusa Slicer I get the following logs:
```
[2024-12-01 13:37:33.917404] [0x00007fb957cb5380] [error]   Failed communication with Prusa Account: 0x561364b59408 action failed (401): {"detail":"Authentication credentials were not provided."}
[2024-12-01 13:37:34.143228] [0x00007fb957cb5380] [error]   Failed communication with Prusa Account: 0x561364b5a788 action failed (400): {
  "message": "key parameter is invalid"
}
[2024-12-01 13:37:34.229541] [0x00007fb957cb5380] [error]   Failed communication with Prusa Account: 0x561364b5a688 action failed (401): {"message": "Missing or invalid security token", "code": "UNAUTHORIZED"}
```

And when I click on "Prusa Connect" at the top I get:
```
[2024-12-01 13:37:33.917404] [0x00007fb957cb5380] [error]   Failed communication with Prusa Account: 0x561364b59408 action failed (401): {"detail":"Authentication credentials were not provided."}
[2024-12-01 13:37:34.143228] [0x00007fb957cb5380] [error]   Failed communication with Prusa Account: 0x561364b5a788 action failed (400): {
  "message": "key parameter is invalid"
}
[2024-12-01 13:37:34.229541] [0x00007fb957cb5380] [error]   Failed communication with Prusa Account: 0x561364b5a688 action failed (401): {"message": "Missing or invalid security token", "code": "UNAUTHORIZED"}
```

When logging out and in again, I get the the same/similar logs as on startup:
```
[2024-12-01 13:42:37.463498] [0x00007fb957cb5380] [error]   WebView runtime error: {"action":"ERROR","error":"{\"status\":401,\"body\":\"{\\\"message\\\": \\\"Missing or invalid security token\\\", \\\"code\\\": \\\"UNAUTHORIZED\\\"}\"}","critical":false}
[2024-12-01 13:42:37.601221] [0x00007fb957cb5380] [error]   Failed communication with Prusa Account: 0x561364b5a788 action failed (400): {
  "message": "key parameter is invalid"
}
[2024-12-01 13:42:37.692269] [0x00007fb957cb5380] [error]   Failed communication with Prusa Account: 0x561364b5a688 action failed (401): {"message": "Missing or invalid security token", "code": "UNAUTHORIZED"}
```

mlbarnes2, 2024-12-01:
Same problem! I am having to log back in several times a day. Using MacOS.

itshanney, 2025-01-01:
I ended up giving up on using the Prusa Connect tab in Slicer and just keep a tab in Chrome open to connect.prusa3d.com, which works flawlessly. Something is just not working right with the web browser component used in the Slicer app and using a regular browser works fine.

jadoglio, 2025-01-12:
This is a really annoying issue.  Why hasn't Prusa support responded to this string.  This need to be fixed.  Also, every time I try to set a nozzle preheat temperature the system crashes and the machine has to be rebooted.

fitrider82, 2025-01-26:
Same issue, rebooting PrusaSlicer resolves the issue but I lose all of my settings, please fix this.

zbkingCZ, 2025-01-26:
I noticed something interesting. Even though PrusaSlicer displays a message about the user being logged out, it is still possible to upload a .BGCODE file to Prusa Connect. See the screenshots below.

1. During the work, a message appeared: “Failed to connect to Prusa Account.”

[image]

2. Despite this, it is still possible to successfully upload a .BGCODE file.

[image]

3. Verification in Prusa Connect through a web browser.

[image]

bryn51, 2025-01-26:
Its pretty clear that the integration between Prusa Slicer and Prusa Connect is broken.

Its not entirely clear why we have this type of interface, with the login button on prusa slicer main screen and something different underlying the "prusa connect" button, which remembers prior login credentials, but forgets when a login timeout occurs. Its a crazy mixed up approach.

Here is a better approach:
a) alter "user preferences" to allow entry and storage of Prusa Connect credentials. If they are left blank then prusa connect button is not displayed on plater screen.
b) disappear the current login function on prusa slicer plater screen.
c) when user opens "prusa connect" the login status is checked and if necessary reauthentication takes place automatically. The goal is to have the PC information displayed ALWAYS valid and up to date. (Thats not the case right now)
d) when user slices the model and "send" button is made visible, when "send" is clicked, authentication takes place seamlessly and automatically.

i see no reason why there need to be two tickets on the same topic, as the same underlying cause is involved.

But for sure Prusa need to take action to complete the task of integrating PC and PS. If the goal is to encourage folks to use PC the current poor effort does not achieve that goal.

> I noticed something interesting. Even though PrusaSlicer displays a message about the user being logged out, it is still possible to upload a .BGCODE file to Prusa Connect. See the screenshots [...]

closed 2026-09-29 by prusaslicer-bot[bot]