prusa3d/PrusaSlicer #13508: Selectable Fuzzy Skin Surface

koloman360, 2024-10-22:
**Is your feature request related to a problem? Please describe.**
I can currently choose all or outside surfaces for fuzzy skin surface application. I cannot apply fuzzy skin to a specific surface.

**Describe the solution you'd like**
Addition of a third choice to the fuzzy skin drop down menu that allow the user to choose a surface or multiple surfaces on which to apply a fuzzy surface.

**Describe how it would work**
Click the new third choice, for example "Choose Surface(s)",  in the fuzzy skin drop down menu. Click the the surface the target surface. By holding down CTRL, click multiple surfaces. Fuzzy skin is applied to selected surface(s). 

**Describe alternatives you've considered**
Mimic fuzzy skin where required in the design of the 3D part.

**Additional context**
I need to 3D print a cylindrical spacer and I want to add fuzzy skin to the inner vertical surface. Thought it would be preferable to have the ability to choose any surface.

neophyl, 2024-10-22:
So basically https://github.com/prusa3d/PrusaSlicer/issues/6988  given that one of the paint options normally available can select surfaces.

koloman360, 2024-10-23:
similar but easier, faster and more accurate

lukasmatena, 2024-11-21:
Implemented in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

closed 2024-11-21 by lukasmatena