prusa3d/PrusaSlicer #13509: Supports don't get generated properly

unhuman, 2024-10-22:
### Description of the bug

https://www.printables.com/model/203613-tifa-lockhart-final-fantasy-7-polygon-model/files

When I try to use organic supports, the hair on the back is not supported.

### Project file & How to reproduce

[tifa.zip](https://github.com/user-attachments/files/17483069/tifa.zip)
[Poly_2_Tifa.zip](https://github.com/user-attachments/files/17483070/Poly_2_Tifa.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 11

### Printer model

Creatality Ender 3 Pro 2

u89djt, 2024-10-23:
(From a fellow user) Hi there, that's the expected behaviour given the specifics of the model you're asking it to deal with. There are no regions on the ponytail shallower than the tolerated overhang angle; just the lowest tip. That tip is sharper than can be made physical with extrusions, so it's missing between the main mass and the tip.
[image]
The grid supports are actually avoiding the ponytail while supporting other parts of the model. Here's what it's actually doing in terms of support:
[image]
You need to paint supports that will hold that tall chunk of independent mass as it's extruded. Maybe something like this:
[image]
[image]
But then you'll see that the tips of the spikey bangs are also unsupported. And then you'll notice that the elbow isn't attached. It's a PS1 graphics model, not a printable model. The face has two extra plates hanging in space, which I would guess are where the eye/nose and mouth textures are supposed to be.
It's not a slicer bug; it's a model that hasn't been crafted to be printable.
I can't lie to you about your chances, but... you have my sympathies.

unhuman, 2024-10-24:
Wow - thank you for the thorough assessment.  Really informative.

closed 2024-10-24 by unhuman

robustini, 2024-11-02:
Another example of media not generated with V2.8.1.
The center hexagon, while creating the base for the support then does not generate it in snug style.

[image]

[TX16S_8db_antenna_plate_V2.zip](https://github.com/user-attachments/files/17606010/TX16S_8db_antenna_plate_V2.zip)

u89djt, 2024-11-02:
@robustini your third party printer profile has an XY separation setting of 1mm, That's why there's a big gap between the object and the support. Look at it from above and you'll see that there's no way for support material to contact the rim you're trying to support. Change it to a more common 80% (literally the characters 8, 0, % to make it a percentage of the extrusion width) and you'll be fine.
[image]