prusa3d/PrusaSlicer #13513: Errors in the slicing process

willythecat, 2024-10-23:
### Description of the bug

After creating a model in sketchup and cleaning the model multiple times, the slicer created holes in the model. 
Happend also with Prusa printers.
[touch 5.2 gehaeuse_bodenc.zip](https://github.com/user-attachments/files/17496240/touch.5.2.gehaeuse_bodenc.zip)

### Project file & How to reproduce

[touch 5.2 gehaeuse_bodenc.zip](https://github.com/user-attachments/files/17496257/touch.5.2.gehaeuse_bodenc.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

windows 11

### Printer model

ender 3

neophyl, 2024-10-23:
Your model is showing multiple errors in prusa slicer
[image]

If I load it into Blender and run its analysis on it that too shows 45 non manifold edges and multiple other things like 63 intersecting faces etc.
For example the non manifold edges which are the biggest issue for slicers are here 
[image]
in yellow.

Using the built in option available on windows 10 to repair using the windows algorithm it then slices ok
[image]

Attached is a fixed version.  
[touch 5.2 gehaeuse_bodenc_fixed.zip](https://github.com/user-attachments/files/17497271/touch.5.2.gehaeuse_bodenc_fixed.zip)

So if you fix your models to make sure they are properly manifold you wont have any issues.  Sketchup is known for producing non manifold model issues.

willythecat, 2024-10-24:
Hello,

great, many thanks for this explanations. I’m using the German version and my question is: How did you repaired the model?
Which tool you used  or where can I find the option in Prusa Slicer?

Many thanks
willythecat

image.png (view on web)<https://github.com/user-attachments/assets/207ea3eb-344a-47f5-8ffe-c7951a5d3027>

image.png (view on web)<https://github.com/user-attachments/assets/8eb14e78-e719-4ccc-aa45-f4253a9085bd>

image.png (view on web)<https://github.com/user-attachments/assets/f961ca6d-a75a-48bf-8f4b-363b6c62e1b7>

neophyl, 2024-10-24:
In windows (10 at least) prusa slicer can access a built in windows repair tool.  See the screen grab above of the model in the tree list, the one with the yellow warning icon informing you of issues.  

Well if you right click the icon it will automatically use that repair tool.  You can also right click the model on the plater and there is a repair option.  

Its not available on Mac/Linux and I don't know about Win11 as I will be sticking with 10 for as long as I can.

willythecat, 2024-10-24:
OK, I found the function in the elements list. Cool, it also worked in Windows 11 latest version.

closed 2026-09-29 by prusaslicer-bot[bot]