prusa3d/PrusaSlicer #13516: Multiple SLA support types in the same file

kozross, 2024-10-23:
**Is your feature request related to a problem? Please describe.**

When supporting resin prints, it is common to use multiple styles of support, varying for thickness, tip diameter and so on. This is a standard practice, [well-documented in guides](https://docs.mango3d.io/doc/j3d-tech-s-guide-to-resin-printing/print-settings/advanced-supporting/), as it allows you to have the most optimal choices in all cases. For example, a single model could easily require three sizes of supports:

* Light supports to hold up critical islands without scarring
* Medium supports to connect the model to its raft
* Heavy supports for internal islands for hollowing

All major SLA slicing solutions have ways of doing this: PrusaSlicer as-current does not. This forces us into a painful decision of either choosing the _lightest_ supports required (potentially risking print failure) or the _heaviest_ required (making scarring almost a certainty, if the supports come off at all). There is currently no workaround for this.

**Describe the solution you'd like**

The ability to manually place supports of multiple settings on the same model, similar to how this is done in Lychee or Chitubox.

**Describe how it would work**

Instead of having a single 'Supports' setting, allow multiple. Then, when manually placing supports using 'SLA Support Points', you should be able to choose which profile is used for which support. Colour-coding them on the model would help as well.

**Describe alternatives you've considered**

Essentially, the current alternatives are:

* Use the lightest type of support you need and risk print failure
* Use the heaviest type of support you need and risk scarring or support irremovability
* Give up and use a different slicer (which is what 99% of people end up doing)

**Additional context**

PrusaSlicer is currently the _only_ open-source SLA slicing solution in existence. It would be very easy for it to eat Lychee, and Chitubox's, _entire_ lunch, but as-current, missing functionality like this really holds it back.

u89djt, 2024-10-24:
From an intrigued FFF user who hasn't tried SLA yet: two things. First, you can set per-object support parameters, but two copies of each are added with half of them greyed out, which seems like a bug, and second, you can do something I can't, which is to save an object with its supports.

(Doubled-up) object modifier additions:
[image]
[image]
[image]
[image]

And this is what it looks like to save a supported object and put that on the plate. I first sliced with defaul supports and saved that, then sliced with the experimental branching supports to get a mixture you can't achieve by using modifiers (I think?). You can disable supports on that imported supported object it so you can experiment with the next object. This is different for SLA than FFF because supports in FFF are extruded differently. In SLA printing it's just a volumetric specification, so an stl can be a complete description of the print job.
[image]
[image]

u89djt, 2024-10-24:
Ah, nope, the duplicated/greyed out support settings are for default and branched supports, with the inactive bank greyed out when you swap the main support settings between default and branched.
I think I probably haven't said anything useful. I'll delete this and the comment above unless someone tells me there's anything relevant in them.

Jony01, 2025-03-03:
You could manualy eddit points and set them size of the head.
[image]

kozross, 2025-03-03:
@Jony01 - thanks for that! That definitely helps. However, this is only a partial solution as far as I can tell: you can vary tip diameter, but not, say, penetration, or thickness of support. Lychee and Chitubox allow that and then some, on the same model, _at placement time_.

Jony01, 2025-03-04:
Could you be more specific (provide some images or 3mf example),
Where does one need to modify penetration for a specific support point interface?
IMHO it is useless.

I do not understand what you mean by "thickness" properly. But I guess you would like to modify the supporting structure to safe resin, which is IMHO risky to edit and fine-tune manually. I see benefits but in comparison with needed skill and effort, one must invest to modify it. There are only a few cases, where it matters: e.g. tunning for mass production, but when you calculate the time spent on the verification, you find that savings are not as high as you expected (there are better optimizations). 
I have to highlight that this is only my point of view.

kozross, 2025-03-04:
It is very typical practice when supporting things like miniatures to have different thicknesses of supports. This has absolutely nothing to do with 'saving material'. I encourage you to read [this resource on supporting miniatures here](https://docs.mango3d.io/doc/j3d-tech-s-guide-to-resin-printing/print-settings/introduction-to-supporting), as this explains quite well why different support sizes on the same model are often needed.

closed 2026-09-29 by prusaslicer-bot[bot]