prusa3d/PrusaSlicer #13518: Rear seam not aligned to rear sharp corner

amatulic, 2024-10-24:
### Description of the bug

This may be related to issue #8841 from 2022 (still open) although that issue is more about seams on curves. This is about a seam that should be on a sharp rear corner. If the cause is the same, please close this as a dupllicate.

When I have a model with a sharp corner, I rotate it on the build plate so the corner is the rear-most feature, and I set the seam position to "rear".

This usually works, but sometimes it doesn't. In this example, the seam is set to "rear" but it appears in three places along the edge, not at the corner that is obviously the rear-most feature on the build plate.

[image]

It doesn't matter if it's Classic or Arachne, it doesn't matter if I set staggered inner seams or external perimeters first, it does the same thing every time.

### Project file & How to reproduce

[boxcoverplate_seamtest.zip](https://github.com/user-attachments/files/17511666/boxcoverplate_seamtest.zip)

The attached project file is for a Prusa MK3S with MMU2S.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 11 - build 22631.4317

### Printer model

Prusa MK3S/MMU2S

SachCZ, 2024-10-25:
Hello, I see the issue. We will try to address that. I have crated a bug report in our system: SPE-2537

amatulic, 2024-10-25:
> Hello, I see the issue. We will try to address that. I have crated a bug report in our system: SPE-2537

Thank you. Maybe fixing this would also fix #8841.

lukasmatena, 2024-11-21:
Fixed in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

closed 2024-11-21 by lukasmatena