prusa3d/PrusaSlicer #13522: Selectable Sliced Info

FrancoGabrielPerez, 2024-10-26:
It would be nice if we could select the text in sliced info, or have a button to copy to clipboard. I have a spreadsheet i use to calculate prices and I find myself manually coping times and material used constantly to make estimates.

**Describe the solution you'd like**
Make the text in sliced info selectable.

**Describe how it would work**
You select the text and then press ctrl+c to copy it.

**Describe alternatives you've considered**
Have a button to copy the info to the clipboard.

**Additional context**
[image]

mix579, 2024-10-26:
I like the idea. Not that I have an immediate need for it but I can certainly see the use of it.

u89djt, 2024-10-26:
That could be output in a comment at the top of the gcode file.

closed 2026-09-29 by prusaslicer-bot[bot]