prusa3d/PrusaSlicer #13526: wrong gcode uploaded to Connect from PrusaSlicer

m10d, 2024-10-27:
### Description of the bug

I opened a new PrusaSlicer window to create a quick flow cal box for my mk4s, recently upgraded & possibly having issues.
[image]

Logged in via Slicer, `Send to Connect` resulted in in this in my connect mk4 page; the gcode is named correctly ("Shape-Box..."), but is clearly the time & preview glyph of a 23hour _helmet_ I had previously sent to my XL.  
[image]

Further, I had clicked in Slicer "skip queue and set ready", but while the printer WAS apparently set to ready, no print started.  Going to send it again, I now see the Mk4 is at ready (previously it was stopped)
[image]

I believe this is a Slicer but, NOT a connect bug, because when I re-send from the same window, it's sending the 30+MB helmet file (which was never opened in this slicer window; I **do** have it open in another window)
[image]
[image]

### Project file & How to reproduce

see descr.  

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1 flatpak

### Operating system

ubuntu 22.04

### Printer model

mk4, and xl5t

kocikdav, 2024-10-29:
Hello, thank you for reporting this issue. Please help us reproduce the issue by answering these questions.
1. The original helmet gcode - it was sliced in different PrusaSlicer window? Was this window still opened while slicing the cube?
2. Can you reproduce this issue again?
3. If yes, when you export the second gcode to a local folder, is it correct gcode or also wrong?

Thank you.

m10d, 2024-10-29:
1. Correct, I had two PrusaSlicer windows open; the gcode it uploaded was another model I had sliced in the other window. I can't guarantee it was the exact most recent one, but it was my model and I had recently sliced it and sent it to my other printer.

2. I reproduced this state multiple times at the time of report, but do not believe I have seen it since.

3. I answered No obviously, but note for this question in the original reproduction it would seem high-confidence the gcode it actually uploaded was indeed wrong based on (A) stated size & preview image, (B) time to upload and (C) reported size on the printer - because the gcode I asked it uploade from the current window was ~10% of the size.

abatkin, 2025-02-14:
I just experienced the same issue...I have two MK4S printers. I wanted to print model A on printer A and model B on printer B.

- Opened two PrusaSlicers, one with model A, one with model B
- Sliced model A in one window and model B in the other window
- Sent model A to printer A
- Filename looked correct (it included the model name and the print time which was 8h55m)
- ...went off to grab some filament for the other printer
- Looked at printer A and saw that it was actually printing model B. The filename was still model A's filename (which is quite distinct from B's model name and included 8h55m in the filename) though the preview is of model B. Actual time remaining is from model B too (~2h17m because it's a much smaller model)

Note that it is impossible for this to be user error: I didn't/couldn't set the filename, and the filename is the correct name based on the model that I wanted to print.

[image]

ahorn42, 2025-02-14:
Hi,

I experienced the same error with an Prusa XL 5T using PrusaSlicer 2.9.0 (PrusaSlicer-2.9.0+flathub.org)

In addition on the XL this lead to a crash of the tool head into the build plate (red 1) and scratched the PEI sheet hardly with the nozzle :( - The printer somehow detected the crash and tried to home (orange path in order 1, 2, 3), but during that process it moved the nozzle over the printbed without lowering the printbed. As this didn't stop I turned the power off and homed it after the powerup. The tool head seems to still work, but I suggest it is not good for the nozzle and heatbreak, if the whole toolhead bends a few mm when pushing agains the printbed! Also the printbed lost its satin surface in the area where the nozzle scratched it.

First I thought I might have missed a step or something, but it now happend the second time. I still can't (and don't really want to :D) reproduce it. To describe my setup: I used multiple instances of prusaslicer in paralell and could upload the first model successfully. Then, when I uploaded the second model it only uploads the code from the first model (and seems somehow to mixup the toolhead settings for those files. File one should be printed with T5 and file 2 with T1).
If I print those files directly from the printer file 1 works as expected and file 2 again crashes the toolhead T5 (which should not be used in file 2 at all)
One thing that came me in mind - may be it is related to putting my computer into standby during uploading the first and second model and something is cached wrongly? In addition, if I change some parameters and slice file 2 again it uplaods the correct model and everything works fine like usually :)

I also secured the second malicious file, if you want, I can upload it somewhere to provide it for analysis unfortunately it is bgcode, so I couln'd have an easy look into it yet.

[image]

[image]

closed 2026-09-29 by prusaslicer-bot[bot]