prusa3d/PrusaSlicer #13527: PrusaSlicer 2.8.1 crashes very often on Arch Linux

BerndAmend, 2024-10-27:
### Description of the bug

PrusaSlicer 2.8.1 crashes very often on Arch Linux.
The crashes occur mostly when changing the infill and re-slicing afterward.
The backtrace I see with coredumpctl was so far the same.

```gdb
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007a21b3013463 in __pthread_kill_internal (threadid=<optimized out>, signo=6) at pthread_kill.c:78
#2  0x00007a21b2fba120 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007a21b2fa14c3 in __GI_abort () at abort.c:79

#4  0x00007a21b42d3af0 in std::__glibcxx_assert_fail (file=file@entry=0x55f368f61b58 "/usr/include/c++/14.2.1/bits/stl_vector.h", line=line@entry=1149, 
    function=function@entry=0x55f368f62868 "std::vector<_Tp, _Alloc>::const_reference std::vector<_Tp, _Alloc>::operator[](size_type) const [with _Tp = double; _Alloc = std::allocator<double>; const_reference = const double&; size_type = long u"..., condition=condition@entry=0x55f3690baac0 "__n < this->size()") at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/assert_fail.cc:41
#5  0x000055f3671b92c4 in std::vector<double, std::allocator<double> >::operator[] (this=<optimized out>, __n=<optimized out>) at /usr/include/c++/14.2.1/bits/stl_vector.h:1149
#6  0x000055f3689fe44f in DoubleSlider::DSForLayers::is_wipe_tower_layer (this=0x55f3abed6760, tick=<optimized out>) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/src/slic3r/GUI/DoubleSliderForLayers.cpp:1043
#7  DoubleSlider::DSForLayers::is_wipe_tower_layer (this=0x55f3abed6760, tick=53) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/src/slic3r/GUI/DoubleSliderForLayers.cpp:1037
#8  DoubleSlider::DSForLayers::get_label(int, DoubleSlider::LabelType, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const::{lambda(int, DoubleSlider::LabelType)#1}::operator()(int, DoubleSlider::LabelType) const [clone .isra.0] (value=value@entry=53, label_type=label_type@entry=DoubleSlider::ltHeightWithLayer, __closure=<optimized out>)
    at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/src/slic3r/GUI/DoubleSliderForLayers.cpp:1110
#9  0x000055f367ea300d in DoubleSlider::DSForLayers::get_label (this=this@entry=0x55f3abed6760, pos=<optimized out>, pos@entry=53, label_type=label_type@entry=DoubleSlider::ltHeightWithLayer, fmt="%1$.2f")
    at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/src/slic3r/GUI/DoubleSliderForLayers.cpp:1132
#10 0x000055f367ea310d in DoubleSlider::DSForLayers::get_label[abi:cxx11](int) const (this=0x55f3abed6760, pos=53) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/src/slic3r/GUI/DoubleSliderForLayers.hpp:191
#11 0x000055f367e80108 in DoubleSlider::Manager<double>::Init(int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)::{lambda(int)#1}::operator()[abi:cxx11](int) const (__closure=<optimized out>, pos=<optimized out>) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/src/slic3r/GUI/ImGuiDoubleSlider.hpp:206
#12 std::__invoke_impl<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, DoubleSlider::Manager<double>::Init(int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)::{lambda(int)#1}&, int>(std::__invoke_other, DoubleSlider::Manager<double>::Init(int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)::{lambda(int)#1}&, int&&) (__f=...) at /usr/include/c++/14.2.1/bits/invoke.h:61
#13 std::__invoke_r<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, DoubleSlider::Manager<double>::Init(int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)::{lambda(int)#1}&, int>(DoubleSlider::Manager<double>::Init(int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)::{lambda(int)#1}&, int&&) (__fn=...) at /usr/include/c++/14.2.1/bits/invoke.h:116
#14 std::_Function_handler<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (int), DoubleSlider::Manager<double>::Init(int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)::{lambda(int)#1}>::_M_invoke(std::_Any_data const&, int&&) (__functor=..., __args#0=<optimized out>) at /usr/include/c++/14.2.1/bits/std_function.h:291
#15 0x000055f367e8c1c8 in std::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >(int)>::operator() (this=<optimized out>, __args#0=<optimized out>, this=<optimized out>, 
    __args#0=<optimized out>) at /usr/include/c++/14.2.1/bits/std_function.h:591
#16 DoubleSlider::ImGuiControl::get_label[abi:cxx11](int) const (this=0x55f3abed6798, pos=53) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/src/slic3r/GUI/ImGuiDoubleSlider.cpp:278
#17 DoubleSlider::ImGuiControl::render (this=this@entry=0x55f3abed6798) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/src/slic3r/GUI/ImGuiDoubleSlider.cpp:694
#18 0x000055f367ea5e91 in DoubleSlider::DSForLayers::Render (this=0x55f3abed6760, canvas_width=<optimized out>, canvas_height=<optimized out>, extra_scale=<optimized out>, offset=<optimized out>)
    at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/src/slic3r/GUI/DoubleSliderForLayers.cpp:982
#19 0x000055f367d21412 in Slic3r::GUI::Preview::render_sliders (this=0x55f3ab75a4d0, canvas=...) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/src/slic3r/GUI/GUI_Preview.cpp:333
#20 Slic3r::GUI::Plater::priv::render_sliders (this=<optimized out>, canvas=...) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/src/slic3r/GUI/Plater.cpp:3359
#21 Slic3r::GUI::Plater::render_sliders (this=<optimized out>, canvas=...) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/src/slic3r/GUI/Plater.cpp:6588
#22 Slic3r::GUI::GLCanvas3D::render (this=0x55f3abed3cc0) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/src/slic3r/GUI/GLCanvas3D.cpp:1959
#23 0x000055f367d2e67c in Slic3r::GUI::GLCanvas3D::_refresh_if_shown_on_screen (this=0x55f3abed3cc0) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/src/slic3r/GUI/GLCanvas3D.cpp:5382
#24 Slic3r::GUI::GLCanvas3D::_refresh_if_shown_on_screen (this=0x55f3abed3cc0) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/src/slic3r/GUI/GLCanvas3D.cpp:5369
#25 0x000055f367cfd1ed in Slic3r::GUI::GLCanvas3D::on_idle (this=0x55f3abed3cc0, evt=...) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/src/slic3r/GUI/GLCanvas3D.cpp:2691
#26 Slic3r::GUI::GLCanvas3D::on_idle (this=0x55f3abed3cc0, evt=...) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/src/slic3r/GUI/GLCanvas3D.cpp:2666
--Type <RET> for more, q to quit, c to continue without paging--
#27 0x000055f3686cdb0c in wxEvtHandler::ProcessEventIfMatchesId (entry=..., handler=<optimized out>, event=...)
    at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1431
#28 0x000055f3686d2cab in wxEvtHandler::SearchDynamicEventTable (this=0x55f3ab799e70, event=...) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1901
#29 wxEvtHandler::TryHereOnly (this=this@entry=0x55f3ab799e70, event=...) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1624
#30 0x000055f3686d8c93 in wxEvtHandler::TryBeforeAndHere (this=0x55f3ab799e70, event=...) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/include/wx/event.h:4007
#31 wxEvtHandler::ProcessEventLocally (this=this@entry=0x55f3ab799e70, event=...) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1561
#32 0x000055f3686d8de2 in wxEvtHandler::ProcessEvent (this=0x55f3ab799e70, event=...) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1534
#33 0x000055f36850f0b4 in wxEvtHandler::SafelyProcessEvent (this=<optimized out>, event=...) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/co [... cut ...]
#54 0x000055f367993c39 in wxEntry (argc=@0x55f3695c3484: 1, argv=<optimized out>) at /usr/src/debug/prusa-slicer/PrusaSlicer-version_2.8.1/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/init.cpp:503
[... 8 lines cut ...]
The issue occurs with every stl-file I tested.
Steps to reproduce:

1. Add a stl-file
2. Slice the project
3. Change slicing parameters e.g. infill
4. Re-slice
5. Repeat until it crashes, sometimes it happens after a couple of minutes, sometimes after hours

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Arch Linux last updated 2024-10-27

### Printer model

Prusa XL

BerndAmend, 2024-11-12:
I use the binaries build by https://gitlab.archlinux.org/archlinux/packaging/packages/prusa-slicer/-/blob/main/PKGBUILD?ref_type=heads

probonopd, 2024-12-04:
@BerndAmend out of interest, does this also happen with the AppImage from https://github.com/prusa3d/PrusaSlicer/issues/13653?

BerndAmend, 2024-12-16:
So far I haven't seen any crashes with the PrusaSlicer 2.9.0-beta1 AppImage.
I got random crashes with the AppImage of PrusaSlicer 2.8.1, but without a stack trace I couldn't tell where they were coming from.

closed 2024-12-16 by BerndAmend