prusa3d/PrusaSlicer #13528: Multimaterial painting limited to 16 extruders and using the 16th crashes the app

NielschB, 2024-10-27:
### Description of the bug

When selecting multimaterial painting on an object, there are only 16 extruders shown while I have 18.
Additionally when using the 16th on the object, when slicing, prusaslicer crashes with "access violation"

### Project file & How to reproduce

Set number of printer extruders to 18, add any object and select multimaterial painting. The extruder list is limited to 16.
When using the 16th on the object and start slicing prusaslicer crashes.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1 Win64

### Operating system

Windows 11 24H2

### Printer model

Custom build, using MK3S as base profile

NielschB, 2024-10-27:
[image]

NielschB, 2024-10-27:
[image]

Jan-Soustruznik, 2024-11-08:
Closed as duplicete of https://github.com/prusa3d/PrusaSlicer/issues/12247

closed 2024-11-08 by Jan-Soustruznik