prusa3d/PrusaSlicer #13530: Telemetry in Prusa Connect lost

Herky3D, 2024-10-27:
### Description of the bug

The telemetry of my Printer MK4S is lost- before dwo days worked normally- show all parameters nice.
Now- nothing. As You can see- printer is PRINTING now.
[image]

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Verze 2.8.1+win64

### Operating system

Edice	Windows 10 Pro Verze	22H2 Datum instalace	‎29.‎07.‎2021 Build operačního systému	19045.5011 Výkon	Windows Feature Experience Pack 1000.19060.1000.0

### Printer model

 Original Prusa MK4S FW:  6.1.3+7898

Jan-Soustruznik, 2024-11-08:
Created internal task no. CONN-580

muesli, 2024-12-29:
Can confirm I have the same issue. Sometimes - very rarely though - it will work all of a sudden.

CloudStrife1983, 2024-12-29:
I confirm this too. Same issue.

Paja-git, 2024-12-29:
I experience the same, is there any progress?

Jan-Soustruznik, 2025-05-16:
Hello @Herky3D, @muesli, @CloudStrife1983, @Paja-git, Could you please check if the issue still occurs? Thank you, and have a great day!

Paja-git, 2025-05-17:
It is OK. It was fixed.

Jan-Soustruznik, 2025-05-26:
@Paja-git Thank you for the information. The ticket has been closed as solved. Have a nice day!

closed 2025-05-26 by Jan-Soustruznik