prusa3d/PrusaSlicer #13532: PrusaSlicer - Change flow control limited by Layer Height/Line Width ratio

adn-art, 2024-10-28:
### Description of the bug

Good morning,
I want to make a print with a layer height of 0.4mm and line width of 2.5mm
However, the slicer limits the flow and therefore the line width which instead of being 2.5mm comes out to 2mm
To find my correct line width the only parameter (that I found) is to increase the layer height to a minimum of 0.6mm
For aesthetic reasons, I really want to stay at a max layer height of 0.4mm, but I would like to have a line width of 2.5mm see for test 3mm How can I disable or change this limiter?
No matter the nozzle diameter entered in the slicer the line thickness is blocked.
I thought: it’s a previewer limitation and the gcode is ok, but i printed to test, result: the display corresponds well to the Gcodethe print was 2mm wide
I also tried with old versions (v2.5.2) of prusa: the problem is still present...
Sincerely

### Project file & How to reproduce

[Debit_G13_Duet_B25_290_hdc04_vit30_L25_D140-64.zip](https://github.com/user-attachments/files/17542939/Debit_G13_Duet_B25_290_hdc04_vit30_L25_D140-64.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

 2.8.1+win64

### Operating system

win 11

### Printer model

cube 200x200x25

rtyr, 2024-10-29:
It's limited only in the preview (you can check E-values in the gcode). Thinner real-life extrusions are likely caused by different issues, not related to slicer.

Tupson444, 2026-09-19:
In my experience, setting large line width causes underextrusion, even when printing slowly (so it is not just max flow rate printer limitation). I think calculation of flow/distance between perimeters is not correct for wide extrusion width.