prusa3d/PrusaSlicer #13536: [insert pause at layer] randomly pauses AFTER layer ?

CaptainFalcon92, 2024-10-29:
### Description of the bug

I've been experiencing an inconvenient problem with the [**insert pause at layer**] feature from PrusaSlicer.

According to [documentation ](https://help.prusa3d.com/article/insert-pause-or-custom-g-code-at-layer_120490)and general observations, it is supposed to pause **BEFORE** the layer seen in the UI.

Instead, the printer very often pauses **AFTER** the chosen layer - and seams especially more likely to do so when using the [**variable layer height**] feature.

Actual result seams output-specific. When experiencing the pause issue; the same model sliced in the same parameters always end up pausing after the layer. But different layers adjustments may pause before.

I've been struggling with pausing for a good couple years now. Pretty much since variable layer height got introduced. 

Has anyone else experienced such a problem ?

Thank you, best.

### Project file & How to reproduce

A part with variable layers height :
[image]

Should pause before printing overhang. Definitely did not.
[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.1+win64-202409181359

### Operating system

Windows 11 23H2

### Printer model

MK3S

u89djt, 2024-10-29:
If you're looking for help (from a user like me, or a dev), post a project file where it happens for you. You said it's random. Don't make us keep trying till it happens :)

CaptainFalcon92, 2024-10-29:
While i was asking for anyone having the same issue, yes so be it. Here is the generated file seen in example above.

[Pause_after_layer_L8620.zip](https://github.com/user-attachments/files/17563795/Pause_after_layer_L8620.zip)

I also did further investigation and can absolutely confirm the M601 command was placed AFTER the targeted layer in the UI - at line 8620.

According to previously shown screenshot, you can see how the pause should have happen before this same layer.

[image]

I'm starting to wonder if the M601 command actually got misplaced due to some height rounding error.

u89djt, 2024-10-30:
Aye, the point to remember really is that the devs know more about what's going on than you or I do, so they tell you what they need in the submission instructions because you absolutely legitimately can't necessarily work it out for yourself. I was just adding motivation that's obvious from the outside perspective. Picture a chain of comments in the submissions where a bunch of people with experience of the problem post "me too!" and a screenshot but no project files. Very frustrating for the devs. The hub is dotted with devs requesting that a submitter post something that's asked for in the guidance, as well as more analysis as you've provided there in your second comment. They're manically busy, so give them what they ask for.
I'm here to find out about problems before they hit me when I'm in a hurry.

u89djt, 2024-10-30:
@CaptainFalcon92 post the *project file*, or if you can't post that one, post another one that exhibits the problem.

CaptainFalcon92, 2024-11-04:
This project file i cannot share due to it's sensible nature. It's a company part.
Here is a different personal project that exhibit the exact same problem minutes ago.

Generated Gcode as m601 at the end of target layer.
Pause happen after the target layer.

[3mf project - pause after layer.zip](https://github.com/user-attachments/files/17618488/pause.after.layer.zip)

CaptainFalcon92, 2024-11-04:
When i reset variable layer heights to it's default 0.05 in the project, the observed pause matches with the corresponding layer. So i can confirm that PrusaSlicer will ignore variable layers heights when adding m601 command.

u89djt, 2024-11-04:
This was a journey.

You've either
specified a pause at 2.15mm above the bed in the constant layer height case, and that translates to 2.151mm above the bed with those specific variable heights you've selected because there's no interface between extrusions coinciding with 2.15 precisely,
or
picked it in the variable view as 2.151 and it translated to 2.15, it doesn't matter in this case.
[image]

We can see in the gcode that a pause is inserted in the variable and constant cases _before_ extrusions start to be laid /onto/ the interface at 2.15mm above the bed.
In the constant case, that extrusion is deposited by the nozzle _from_ 2.243mm above the bed, and in the variable case, it's from 2.2mm above the bed. The _underside_ of the plastic extrusion executed after the pause is at 2.15(1)mm above the bed

In the variable case, the bridging extrusion _layer_ (numbered) appears immediately below the pause _level_ (mm above bed).
In the constant case, the bridging extrusions appear two layers down, with another solid layer on top because the object fills between those regions, which are about half as high as in the variable case.
(Very generally speaking, apart from thin walls and other bits and bobs, plastic is extruded when the model occupies 50% or more of the volume that it will occupy.)

The pause happens where you asked for it, so I assume you thought you were asking for something else.

If you were going off what's said here:
https://help.prusa3d.com/article/insert-pause-or-custom-g-code-at-layer_120490 
then perhaps I can understand why it's different to what you expected.
I suspect that
"Select Add custom G-code
  - The custom G-code will be inserted before the selected layer is being printed"

Should say
 - "The… will be inserted before printing proceeds on top of the current layer"

Time for some input from a dev...

closed 2026-09-29 by prusaslicer-bot[bot]