prusa3d/PrusaSlicer #13537: Crash on MacOS moving the PrusaSlicer window to another monitor

starbuck93, 2024-10-29:
### Description of the bug

PrusaSlicer seems to crash if I want to move the window to an external monitor on MacOS 15. It will open and run just fine on my Macbook monitor but as soon as I begin to move it away, this screenshotted error pops up which says "no bitmap handler for type 17 defined."

### Project file & How to reproduce

[image]

[blade-5-vase.3mf.zip](https://github.com/user-attachments/files/17560332/blade-5-vase.3mf.zip)

The MacOS crash screen had these details:

```
Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   PrusaSlicer                   	       0x10150aafc Slic3r::GUI::Plater::get_notification_manager() + 4
1   PrusaSlicer                   	       0x10132b7c8 Slic3r::GUI::GLCanvas3D::_set_warning_notification(Slic3r::GUI::GLCanvas3D::EWarning, bool) + 492
2   PrusaSlicer                   	       0x10132af1c Slic3r::GUI::GLCanvas3D::~GLCanvas3D() + 24
3   PrusaSlicer                   	       0x1014854f4 Slic3r::GUI::View3D::~View3D() + 40
4   PrusaSlicer                   	       0x1030992c4 wxWindowBase::DestroyChildren() + 152
5   PrusaSlicer                   	       0x103115168 wxWindow::~wxWindow() + 940
6   PrusaSlicer                   	       0x10151efd0 Slic3r::GUI::Plater::~Plater() + 12
7   PrusaSlicer                   	       0x1030992c4 wxWindowBase::DestroyChildren() + 152
8   PrusaSlicer                   	       0x103115168 wxWindow::~wxWindow() + 940
9   PrusaSlicer                   	       0x1014dbf44 wxChoicebook::~wxChoicebook() + 192
10  PrusaSlicer                   	       0x1030992c4 wxWindowBase::DestroyChildren() + 152
11  PrusaSlicer                   	       0x10310fe00 wxNonOwnedWindow::~wxNonOwnedWindow() + 92
12  PrusaSlicer                   	       0x1014db190 Slic3r::GUI::MainFrame::~MainFrame() + 12
13  PrusaSlicer                   	       0x102fc5f18 wxAppBase::CleanUp() + 120
14  PrusaSlicer                   	       0x10314e418 wxApp::CleanUp() + 56
15  PrusaSlicer                   	       0x10322d27c wxEntryCleanup() + 68
16  PrusaSlicer                   	       0x10322d3e4 wxEntry(int&, wchar_t**) + 136
17  PrusaSlicer                   	       0x101483b80 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 284
18  PrusaSlicer                   	       0x10091b8b0 Slic3r::CLI::run(int, char**) + 20984
19  PrusaSlicer                   	       0x1009212c4 main + 220
20  dyld                          	       0x195660274 start + 2840
```

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.1+MacOS-x64-202409181403

### Operating system

MacOS 15.1

### Printer model

Prusa Mini

emelymw, 2025-01-17:
i have exactly the same problem with macos 15.1.1 with the prusaslicer 2.9.0

numbakrrunch, 2025-01-17:
Same here.  Sequoia 15.2, prusaslicer 2.9.0

starbuck93, 2025-01-17:
Strangely enough, I cannot reproduce it anymore with macOS 15.2 and PrusaSlicer 2.9.0. I can move a loaded PrusaSlicer window between my laptop display and the external monitor.

PrusaSlicer
Version:   2.9.0
Build:     PrusaSlicer-2.9.0

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 15.2 (Build 24C101)
Total RAM size [MB]: 17,180MB
OpenGL installation
GL version:   4.1.0 (4.1 Metal - 89.3)
Profile:      Core
Vendor:       Apple
Renderer:     Apple M1 Pro
GLSL version: 4.10.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_APPLE_client_storage
GL_APPLE_container_object_shareable
GL_APPLE_flush_render
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_texture_range
GL_ARB_ES2_compatibility
GL_ARB_blend_func_extended
GL_ARB_draw_buffers_blend
GL_ARB_draw_indirect
GL_ARB_explicit_attrib_location
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_occlusion_query2
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_subroutine
GL_ARB_shading_language_include
GL_ARB_tessellation_shader
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_cube_map_array
GL_ARB_texture_gather
GL_ARB_texture_query_lod
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_swizzle
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_texture_compression_s3tc
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_sRGB_decode
GL_NV_texture_barrier
</details>

janchlebek, 2025-01-19:
Hi everyone, for me this started to happen yesterday - PrusaSlicer opens just fine, I can use it on my laptop screen but as soon as I try to move it to external display it crashes with the above error.

janchlebek, 2025-01-19:
Since the error is complaining about inaccessible file I tried to put a random small JPG file in that location with name mentioned in the error. From that point I can move PS to the other screen so it can be considered a workaround.

closed 2026-09-29 by prusaslicer-bot[bot]