prusa3d/PrusaSlicer #13539: MK4S + MMU3 - Wipe Tower included in Skirt Draft Shield (PS 2.8.1)

Mike-FUT, 2024-10-29:
### Description of the bug

I sometimes use the Skirt Draft Shield for warping sensitive materials (works as expected) and today first time I used it with multi material print and I noticed that it includes the wipe tower.
This doesn't make any sense since I believe no-one cares if the wipe tower is warping and also it reduces (or nearly eliminates) the effect of the draft shield on the print because of the space between wipe tower and item to be printed, especially when it has to be placed further away to use the "No Sparse Layers (EXPERIMANTAL)" option (which I always do to reduce waste and to speed up).

### Project file & How to reproduce

1. Put any object on the slicer's print bed and make a MMU print (multi color)
2. Add skirt

This will include wipe tower instead of ignoring it with the skirt.
In case this is not a bug and really by design for whatever reason: Please add a checkbox somewhere to exclude wipe tower.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 11

### Printer model

MK4S + MMU3 6.1.3

Mike-FUT, 2024-10-29:
[image]

rtyr, 2024-11-07:
It may not make sense in your particular case with experimental wipe tower or with other specific cases, but I would certainly not say that wipe tower is irrelevant when it comes to warping. It is often quite the opposite.
Wipe tower can be heavily filled based on number of toolchanges, making it even more prone to warping than main object. Warped wipe tower can easily cause fail of the whole print.

closed 2026-09-29 by prusaslicer-bot[bot]