prusa3d/PrusaSlicer #13541: Organic Supports not supported - Suports have unsupported overhangs

anatwick, 2024-10-30:
### Description of the bug

When using  Organic supports prints several prints supports were failing and upon further inspection and zooming into the sliced preview it appears the issue is actually in the slicer.  As you can see from the supports they shift over and are a complete overhang with zero support material below thus causing the issue.   Here is an image clearly showing the problem from the underside: 
[image]

### Project file & How to reproduce

[image]
These are my current support settings but i've changed these settings up thinking they were the problem to no avail. 

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 11

### Printer model

Railcore 2 ZLT

neophyl, 2024-10-31:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/10725  The devs are working on it according to a comment in one of the many closed duplicates.

u89djt, 2024-10-31:
Zip the project file to post it.

Jan-Soustruznik, 2024-11-08:
Closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/10725

closed 2024-11-08 by Jan-Soustruznik