prusa3d/PrusaSlicer #13544: Cut tool - Multiple dovetails

fursphere, 2024-10-31:
**Is your feature request related to a problem? Please describe.**
When using the cut tool, only a single dovetail joint is supported.

**Describe the solution you'd like**
Add the ability to do multiple dovetail joints.  Possibly even support the ability to move them to desired location.

**Describe how it would work**
When cut - dovetail is selected, add a button set that supports "add additional joint" "remove joint"

**Describe alternatives you've considered**
There really isn't an alternative to dovetails. 

**Additional context**
Idea came from the astromech builders, trying to cut large pieces down to smaller printers.  Currently a CAD program must be used (Fusion for example) to add multiple dovetails.  Would be nice to do it right in the slicer.   

thanks!

neophyl, 2024-10-31:
Please search for existing issues before opening new ones.  This is a duplicate, for example -
https://github.com/prusa3d/PrusaSlicer/issues/12655 or
https://github.com/prusa3d/PrusaSlicer/issues/12077

Jan-Soustruznik, 2025-01-09:
closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/12077

closed 2025-01-09 by Jan-Soustruznik