prusa3d/PrusaSlicer #13545: Purge line with same color as first used filament

EmilBjorn, 2024-11-01:
**Is your feature request related to a problem? Please describe.**
When printing with multiple colors, and having chosen one that is not the color of the first "extruder", the purge line will be of the first color, and then immediately issuing an M600 command to change to the correct color. This is annoying when doing manual color changes, and superflous

Current workaround is either recoloring the first color to the wanted color and redoing the colormapping OR manually editing the gcode after slicing, to remove the first M600 in the beginning. This solution complicates transfer to PrusaLink / Prusa Connect, as it can no longer be done exclusively from the slicer.

**Describe the solution you'd like**
Confirm with user that correct color is in nozzle when starting print. And then prime with the color used for the first extrusions on the first layer, without the need for a superflous M600 command.

rtyr, 2024-11-01:
As I understand it, you are reffering to "M600 hack", where you are using M600 command in toolchange script.

> Current workaround is either recoloring the first color to the wanted color and redoing the colormapping OR manually editing the gcode after slicing, to remove the first M600 in the beginning. 

Use `{if previous_extruder!=-1}M600{endif}` instead of just `M600` in toolchange script. It will solve your issue.

rtyr, 2024-11-04:
Answered. No response. Closing.

closed 2024-11-04 by rtyr