prusa3d/PrusaSlicer #13547: Painting a model changes a bridging layer to not a bridge.

CoookieTM, 2024-11-01:
### Description of the bug

I painted a model where a bridge should be. The model is one solid object. After painting it, it is no longer treated as a bridge. It fails to print this way because it ignores the bridge. Produced in PS 2.8.1 and PS 2.7.2
[image]
[City Bus 40 Rigid (1).zip](https://github.com/user-attachments/files/17604623/City.Bus.40.Rigid.1.zip)

### Project file & How to reproduce

1. Paint underside window of bus.
2. Observe layer is no longer a bridge

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 10 Home

### Printer model

Prusa XL

u89djt, 2024-11-02:
(From a fellow user following the hub)
That's a paid model that specifies no derivatives, so I won't post parts of the mesh not necessary for explanation of slicer functions.
https://cults3d.com/en/3d-model/game/city-bus-abbymath

When you paint a region, it becomes a separate part and is printed as such, because an extrusion can't change colour while it's proceeding.
That separate part is constructed by following the paint into the object perpendicular to the surface. This means that the separate window you've painted no longer sits on the perimeters around it, so it doesn't comprise bridges any more.
Painting round the corner a bit doesn't help because currently the painting regime doesn't interpolate round a sharp corner. I would imagine that's on the devs' list somewhere.

The instructions on the object explain that it can be printed in slices. That looks like this if you separate the print into parts for different extruders in prusaslicer:
[image]

To recolour just the windows, I've placed modifiers within the mass of the object to protrude just outside the windows so they're rendered in the second colour. It's fiddly, but that's your only option without redesigning the object, so you'd have to start from scratch.
[modifiers.zip](https://github.com/user-attachments/files/17607278/modifiers.zip)
[image]
[image]

I have deleted the original object. I believe you should delete the whole object project file from your post and maybe replace it with a cut down version that just shows the problem you have. Although I think you can just delete it at this stage, as we know what's happening.

CoookieTM, 2024-11-02:
Yeah I've done the workaround for other models, but this issue keeps coming up. I think when more people use multicolor painting on the XL this will be a stumbling block. You are right it is no distribute. I have removed the project. However, it is also not a paid model on all platforms, as it is free on printables.

u89djt, 2024-11-02:
Rambling:
I guess you'd be looking for a button saying "expand inverted areas of enclosed colour". That could work for the fully painted windows in this case with a frame around them. But picture a dot of colour painted in the middle of one of the lower windows. That doesn't have any good solution that doesn't involve supports, I think.
So you'd need supports - or a warning - for novel parts created by painting with no vertical engagement. You can paint those on, which seems a shame on a model originally designed to survive by bridging, but you're asking more of the model than was intended.
Perhaps we won't see automated solution to things like this till the slicing process itself is interactive and branching. It obviously has to go that way - it's just going to be a lot of work.
The right solution here and now is presumably for designers of objects that play close to the wind with printing orientation to part out things that could be different colours in a way that will print well. That wouldn't be difficult. I guess if it was difficult, you wouldn't expect an automated system to achieve it.

rednaxal, 2024-11-03:
Actually this is a problem for graphics/icons printed on the bottom of models. Very often we do have a print (for example a miniature or a statue) which needs an icon/text at the bottom of it. The nice approach is to make that text/icon's background of an other color. Because of the issue described here the quality of that text/icon usually is very bad. And it is not a problem of a model, I think. It is a purely slicing issue. If only we could force such "flat planes" objects to somehow inject into the base model - that would allow them to be printed as bridges still.

u89djt, 2024-11-03:
@rednaxal can you post an example project file with that problem? If there's copyright to respect in the figure above the problem, just cut it off.

rednaxal, 2024-11-03:
@u89djt Sure. Attaching it here. These are some models for a "Dune - The Battle for Arrakis" board game.
[Dune.zip](https://github.com/user-attachments/files/17610162/Dune.zip)

u89djt, 2024-11-03:
Great, I'll contemplate the orb for a bit and drop in anything that seems pertinent.

u89djt, 2024-11-03:
You could do this:
[planed icon added Dune.zip](https://github.com/user-attachments/files/17611879/planed.icon.added.Dune.zip)
[image]
[image]
That's a common surface for all the painted regions cut back by 0.01mm and expanded internally, then auto painted.
More ideas and impressions available.

rednaxal, 2024-11-03:
@u89djt Yes, that is a good trick that does the job in this particular case. But unfortunately not everyone is experienced in 3d modeling to be able to do this with every model that needs this. :) All these models in my example were taken from other sources (cults3d, printables). And for example I don't have much experience in 3d models editing (especially when a model consists of a set of complex surfaces and not a simple solid object, which is quite often happens in public sources). So it would be good to have some kind of a slicer's setting that allows to "expand" colored surface automatically, to bond it into the base model better.

u89djt, 2024-11-03:
Ah, I assumed those were yours. Are they freely distributable? If not, we should delete the 3mf files.
I would imagine that it's a feasible task if the colour regions are coplanar and have uninterrupted mass above and around them. Being a crusty old engineer, all I'm coming up with so far is pathological cases. :)
For all we know, it could be being tested right now. They're resourceful folk.

rednaxal, 2024-11-03:
Yeah, these are publicly available models on printables.com. So it should be ok.

u89djt, 2024-11-04:
The license specifies attribution, so we are obliged to do that.
https://www.printables.com/model/1046355-dune-war-for-arrakis-retail-insert-custom-componen

closed 2026-09-29 by prusaslicer-bot[bot]