prusa3d/PrusaSlicer #13549: Connection failed Something went wrong.

Herky3D, 2024-11-02:
### Description of the bug

There is annoucement in Prusa Slicer tab after sending file via Prusa connect button - "Connection failed
Something went wrong" but the file is normally start to print..
Looks like Prusa Connect is unconnected after some time.. that is strange for this time very consuption process.
Is there any "time unconnect" function?

### Project file & How to reproduce

Version: 2.8.1+win64
Build: PrusaSlicer-2.8.1+win64-202409181359
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (sestavení 19045), 64bitová edice
Total RAM size [MB]: 12,822MB

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version: 2.8.1+win64 Build: PrusaSlicer-2.8.1+win64-202409181359

### Operating system

Operating System: Windows System Architecture: 64 bit Windows Version: Windows 10 (sestavení 19045), 64bitová edice Total RAM size [MB]: 12,822MB

### Printer model

Original Prusa MK4S

JeppeJonHolt, 2024-11-18:
Go into C:\Users\username\AppData\Local and delete prusaSlicer. This fixed the issue for me

Herky3D, 2024-11-24:
> Go into C:\Users\username\AppData\Local and delete PrusaSlicer. Then just launch PrusaSlicer, this fixed the issue for me

Thanks! This delete didn´t help alone, but after reinstall works nice.

closed 2026-09-29 by prusaslicer-bot[bot]