prusa3d/PrusaSlicer #13550: Issue with complete objects mode along with skirt and brim

JohnKentJr, 2024-11-02:
### Description of the bug

Using the complete individual objects mode with skirt and brim turned on results in the printer first printing the skirt and brim of all objects at once before proceeding to complete the first object. It is expected that the brim and skirt of the other objects is laid down at the same time as the first layer of the object.

### Project file & How to reproduce

Turn on complete individual objects mode, turn on skirt and brim, have more than 1 object in the platter. export and then print. The printer will first lay the entire first layer of skirt and brim for all objects. Then complete the first object. 
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

windows 2.8.1

### Operating system

Windows 10.0.19045

### Printer model

MK3S+

u89djt, 2024-11-02:
(Fellow use following the hub)
The skirt has to be printed first because it isn't generated per object.
Printing the brims at the same time as each object would increase the volume that must be left unobstructed around every object.
You'd be asking for a special case that doesn't seem to be beneficial - is there are reason to print the brims each time?
[image]

JohnKentJr, 2024-11-02:
Hi Dave, I understand this is a bit of an edge case and likely not going to impact many people other than myself. I’ve been making these clocks for a few years now and have found that the use of the brims results in significantly higher precision parts than without. Normally I do not use the individual complete parts options because it takes considerably longer to produce all the parts I need but this time I wanted to try it out because I’m always trying to improve the quality of my product and a reduction in travel between parts was the next step. There are more than 100 parts so use of the complete items option to have as many parts on one bed as possible is necessary.In the current use case I’m working with a 0.15mm nozzle and printing fine tooth gears that without the inner and outer brim often suffer adhesion problems. Not that the piece lifts from the bed but that as they cool the layer contracts slightly and so the first few layers of the gear tooth are ever so slightly tighter than the middle sections.  The precision of those layers is a significant factor in the amount of friction reducing that friction as much as possible is important to the project. Using the brim and a tighter gap between it and the part results in a perfect gear each time because that outer most layer of the brim is the one that contracts but the layers of the part do not. I remove those brims carefully by hand later and the result is worth the effort. In the way that this is currently set up the temperate difference between the brim and the layer of the part after that first part is completed will be drastically different and so it’s going to likely result in a failed part. -John Sent from my iPhone

image.png (view on web)

u89djt, 2024-11-02:
Interesting - it sounds like it could be worth your while cadding bespoke brim objects to add as parts to each component? You can modify the extrusion characteristics to mimic the automatic brim, and could specify the gaps and adhesions very precisely with your choice of geometries.

closed 2026-09-29 by prusaslicer-bot[bot]