prusa3d/PrusaSlicer #13552: G-code Viewer breaks when "; generated by PrusaSlicer xx" line is missing in the file

Impact123, 2024-11-03:
### Description of the bug

When the `; generated by PrusaSlicer xx` is missing the G-code viewer breaks
## With `;generated by PrusaSlicer xx`
[image]
## Without
[image]

### Project file & How to reproduce

1. Export Gcode as normal
2. Open it in G-code viewer and observe it looking as expected
3. Copy file
4. Remove `; generated by PrusaSlicer xx` line form the file
5. Open copy in G-code viewer and compare with `.2`

Alternatively compare and check with these two gcode files.
[reproduction.zip](https://github.com/user-attachments/files/17609383/reproduction.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 11

### Printer model

Gcode generated with default settings for for MK4S@0.4mm

u89djt, 2024-11-03:
(fellow user following the hub)
The "generated by Prusaslicer" and version commented declaration tells the viewer to look for the prusa extrusion type declarations in comments and colour accordingly.
Without that declaration, it doesn't, because it shouldn't assume that it was generated by one of the prusaslicer versions it can interpret.
The viewer doesn't promise to deal with other slicers' bespoke attributes and assertion formats. They could, for example, change without warning and actually break the viewer, or cause glitches that mislead the user.
The view in your case isn't broken, it just doesn't show any prusa extrusion type colouring.
If it isn't a prusaslicer file, the view is of the printing actions per se without abstract information.
This isn't an issue and can be closed.

Impact123, 2024-11-03:
Thanks for thr explanation, I guess that makes some sense. Do you happen to know why the bed shows those "zigzags"? 
It might be a good idea to have a note about this behavior in the viewer when a non prusa G-code file is opened.

u89djt, 2024-11-03:
That's an indication that the distance of bed surface off to those sides is unknown. It's a fairly standard way of indicating there's potentially more not shown. You haven't told it to look for prusa information, so it doesn't interpret the printer specification. The origin is known, so the edges to that are definite.

SachCZ, 2024-11-04:
@u89djt Thank you for the great and accurate explanation. Your great work on this platform is not going unnoticed and we cannot thank you enough. Closing the issue as it is not a bug.

closed 2024-11-04 by SachCZ

u89djt, 2025-05-26:
@sachz don't hesitate to correct errors that I make. I won't be offended or ashamed - the plan is to save you time overall and to learn stuff, and I know my hit rate is fine. For example, when I said in a recent report that the slicer never performs boolean operations, I'd missed that being introduced in 2.6.0. I'm more interested in understanding being established asap than saving face :) Finding and correcting errors is always good. Cheers!