prusa3d/PrusaSlicer #13553: starts printing layer in mid air

koloman360, 2024-11-03:
### Description of the bug

Printing of some layers, which contain bridge(s), start in mid air. 

This bug has existed and has been reported many times in older versions of PrusaSlicer. Still exists in the latest version.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.1+win64-202409181359

### Operating system

Windows 7 (build 7601, Service Pack 1), 64-bit edition

### Printer model

Prusa I3 MK3S+

u89djt, 2024-11-03:
You need to zip the project file to post it.

u89djt, 2024-11-03:
You could do this:
[planed icon added Dune.zip](https://github.com/user-attachments/files/17611822/planed.icon.added.Dune.zip)
[image]
[image]
That's a common face for all the painted areas cut back from the body internally by 0.01mm, auto painted.
More ideas and impressions available.

jmerc77, 2024-11-04:
fyi win7 is so old it may be a compatibility problem. try on win10 or win11 (10 is more stable in my opinion) to see if that helps.

u89djt, 2024-11-04:
@jmerc77 the operating system has no influence on the slicing algorithm.

u89djt, 2024-11-04:
(I'm a fellow user following the hub)
Yep, you'd call that a failure of the arachne algorithm to extrude a bridge between existing extrusions.
In this case, the classic algorithm yields plausible extrusions.
The problem is that you're asking it to render a curved sharp edge just touching into the layer. The quality of the mesh itself could be a factor, but I'd speculate that the problem is inferring continuity on so little information. You can see that it misses:
[image]

If you want a reliable slice under general conditions, you could modify the base model to resemble feasible extrusions more closely. In this case, a small chamfer works. 0.3mm is just what I tried first and it came out OK.
[03chamfer added SW Spacer RevB.zip](https://github.com/user-attachments/files/17617709/03chamfer.added.SW.Spacer.RevB.zip)

[image]
[image]

koloman360, 2024-11-04:
@u89djt Thank you for the suggestions.

 I have an even easier hack, I use the Cura slicer.

Really just hope they  fix the bug... surprised it's still around.

u89djt, 2024-11-04:
Ah right, does the arachne implementation in cura see through it? Can you show that layer? It's got to be interesting to someone.

u89djt, 2024-11-04:
I think that won't fare any better. There's still an extrusion mid-air - that point will droop, so they're unlikely to meet? But it's different, at least.
[image]
[image]

u89djt, 2024-11-04:
Yeah, the classic perimeter generator doesn't try to start an extrusion over nothing.
[image]

koloman360, 2024-11-05:
I'm okay with and except a little droop. With Cura, I don't consider the material to be in "mid air" because it is continuous extrusion and the unsupported span is relatively small. PrusaSlicer  makes the printer create a blob where it tries to print in mid air, then smears that blob elsewhere on the layer. The next layer droops even more because the support PrusaSlicer thinks it has below is not there making the bridge span longer.  

IMHO, PrusaSlicer fares far worse than Cura in this case.  The will be my last time posting about it. PrusaSlicer coders clearly don't seem motivated to fix it.

timur-tabi, 2025-02-20:
I just encountered this bug.  I'm running v2.9.0 on Ubuntu 24.

[image]

[Clamp top Ramped.zip](https://github.com/user-attachments/files/18880747/Clamp.top.Ramped.zip)

timur-tabi, 2025-02-23:
I got another one, and this time it's not the model's fault.

[image]

[usbadustcoverextended-grip.zip](https://github.com/user-attachments/files/18927796/usbadustcoverextended-grip.zip)

u89djt, 2025-02-23:
(From a fellow user) You took responsibility for placing supports:
[image]
Here's the auto support for that scenario:
[image]
See if you can come up with an extrusion scheme that would survive this:
[image]
I think it would have to be a solid raft of bridge extrusions across between the support material lines.

jmerc77, 2025-02-23:
Hmm. I just had a thought. Check the extrusion width of the floating bit and adjust the settings for arachne such that it won't be able to extrude that width. In theory, that awkward fragment should become unslicable instead of unprintable, thus non-existent. The caveat? Other parts of the model may be affected in a slightly negative way.

timur-tabi, 2025-03-28:
Got another one.  This time, I'm not using any supports.  There is a hollow cylinder that starts filling in via the perimeter instead of bridge:

[image]

here's the final layer:

[image]

PS should be able to start the bridge along the outside before trying to print the perimeter of the cylinder.

[magnet_bar.zip](https://github.com/user-attachments/files/19497781/magnet_bar.zip)

timur-tabi, 2026-09-03:
This bug should still be fixed.