prusa3d/PrusaSlicer #13554: Renaming a filament preset with parentheses causes incompatible profiles to appear

jerrygrey, 2024-11-04:
### Description of the bug

When renaming a filament preset to include parentheses "()" in the name causes incompatible profiles to appear. Reloading PrusaSlicer will resolve the issue.

[image]
[image]

### Project file & How to reproduce

Rename filament preset to include parentheses, such as "Fillamentum PLA (MK4)"

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.1+MacOS-x64

### Operating system

MacOS 14.4.1 (23E224)

### Printer model

Prusa MK4

YuSanka, 2024-11-07:
Thanks for your report.
SPE-2556

lukasmatena, 2024-11-21:
Fixed in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

closed 2024-11-21 by lukasmatena