prusa3d/PrusaSlicer #13555: better seams by using a false seam forcing the real one to inside

jmerc77, 2024-11-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
seems are often ugly. this is usually caused by over extrusions sharp during deceleration.

**Describe the solution you'd like**
the idea is to take the 2 outer most perimeters and make them one with its seam inside the part. 

**Describe how it would work**
the tool path will have 2 u turns touching as a false seem. example:
[image]

ignore the inner most seam. it wont be there in the real version as this is an stl trick for example proposes.

as you can see, the outer false seam is staggered. it is not a good idea to align this kind because it may be a weaker seam type.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

i reviewed a fair amount of suggested seams in the requests here and they are either too fiddly to calibrate, copy cats of other slicers, or repeated posts. as far as i can tell, this is new and something easy to try. and it doesnt clone any IP (not even mine as i'm not selling nor copy writing this).

**Additional context**
Add any other context or paint drawing about the feature request here.

something to consider is gap values for the false seam. in fact, adjustments for inner and outer gap separately would be nice.

files from demo (stl, 3mf, scad, the picture above as png):
[3dp false seam example.zip](https://github.com/user-attachments/files/17613110/3dp.false.seam.example.zip)

ramosglauco, 2024-11-04:
Really a great idea. I suggest you also make this suggestion on the Orcaslicer github. The chances of your idea being uploaded there are certainly much greater.

rtyr, 2024-11-04:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/7349 and https://github.com/prusa3d/PrusaSlicer/issues/3328 (and other linked)

jmerc77, 2024-11-04:
hmmm... just thought of another improvement that makes sharper corners in the gap... deliberately under extrude a tiny length before turning the corner. reduces the bulge further.

jmerc77, 2026-09-19:
Any plans for implementation? In theory it may need less tuning than scarf
seems.

Jon