prusa3d/PrusaSlicer #13558: Some models have errors in Gcode generation

yayo119, 2024-11-05:
### Description of the bug

During the printing process of a model, Klipper automatically exited twice at the same position, displaying an error indicating it was out of the printing range with some strange values (30.840, -40.390, 15.000). Upon checking the PrusaSlicer-generated Gcode, I noticed that in the final layers of the model, there were large negative Y values repeatedly appearing.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.1

### Operating system

Version:   2.8.1+MacOS-x64 Build:     PrusaSlicer-2.8.1

### Printer model

air filter

Jan-Soustruznik, 2024-11-11:
Hello, @yayo119 , could you share your project file (3mf). Thank you, have a nice day

Jan-Soustruznik, 2024-11-22:
Hello, @yayo119. Your issue appears to be similar to [#13478](https://github.com/prusa3d/PrusaSlicer/issues/13478). This problem should be resolved in the latest alpha release (2.9.1), https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1

Please check if the issue persists with this version. For now, I will close this issue as a duplicate. However, if you still encounter the same problem with the latest release on any 3MF file, please create a new issue and share your project file.

Thank you for your report, and have a great day!

closed 2024-11-22 by Jan-Soustruznik